gazebosim/gz-common #65: Image::MaxColor scales very poorly with image size

scpeters, 2020-05-21:
While debugging why large Image heightmaps are so slow to load in osrf/gazebo, I noticed that the loader was spending lots of time in `Image::MaxColor` (via Heightmap::MaxElevation). I've added a performance test in [branch scpeters:image_max_color_performance](https://github.com/ignitionrobotics/ign-common/compare/ign-common3...scpeters:image_max_color_performance?expand=1), with square images of increasing size 2^N+1 x  2^N+1 (129x129, 257x257, 513x513, 1025x1025) added in https://github.com/ignitionrobotics/ign-common/commit/02ffa909169b34fda32115e041afe3bb4b7ed9b2 and a performance test added in d3eb55b. The images increase in size by roughly a factor of 4, but the time spent calling `Image::MaxColor` in the performance test increases by factors greater than 10 in my first tests on my mac laptop:

~~~
[----------] 4 tests from FlatHeightmaps/ImagePerformanceTest
[ RUN      ] FlatHeightmaps/ImagePerformanceTest.MaxColorFlatHeightmap/0
Error opening log file: /Users/scpeters/tmp-ign//test_logs/FlatHeightmaps/ImagePerformanceTest_MaxColorFlatHeightmap/0.log
[       OK ] FlatHeightmaps/ImagePerformanceTest.MaxColorFlatHeightmap/0 (261 ms)
[ RUN      ] FlatHeightmaps/ImagePerformanceTest.MaxColorFlatHeightmap/1
Error opening log file: /Users/scpeters/tmp-ign//test_logs/FlatHeightmaps/ImagePerformanceTest_MaxColorFlatHeightmap/1.log
[       OK ] FlatHeightmaps/ImagePerformanceTest.MaxColorFlatHeightmap/1 (3551 ms)
[ RUN      ] FlatHeightmaps/ImagePerformanceTest.MaxColorFlatHeightmap/2
Error opening log file: /Users/scpeters/tmp-ign//test_logs/FlatHeightmaps/ImagePerformanceTest_MaxColorFlatHeightmap/2.log
[       OK ] FlatHeightmaps/ImagePerformanceTest.MaxColorFlatHeightmap/2 (49867 ms)
[ RUN      ] FlatHeightmaps/ImagePerformanceTest.MaxColorFlatHeightmap/3
Error opening log file: /Users/scpeters/tmp-ign//test_logs/FlatHeightmaps/ImagePerformanceTest_MaxColorFlatHeightmap/3.log
[       OK ] FlatHeightmaps/ImagePerformanceTest.MaxColorFlatHeightmap/3 (764936 ms)
[----------] 4 tests from FlatHeightmaps/ImagePerformanceTest (818616 ms total)
~~~

* 129x129: 261 ms
* 257x257: 3551 ms
* 513x513: 49867 ms
* 1025x1025: 764936 ms

We should be able to speed this up significantly, which will be helpful for reducing the time spent loading Heightmaps.

mjcarroll, 2020-05-21:
Just taking a quick look, I would expect it to grow `N^2`, because iterating over every pixel is unavoidable.  I think that the `Image::Pixel` call is probably hurting us a pretty good bit, since we are building a struct to return for that call every time.  Also, it looks like that `FreeImage_GetPixelColor` is doing bounds checking, which is also slow.

I did a quick test in Python with OpenCV and got:
```
129 - 10000 loops, best of 3: 29.3 µs per loop
257 - 10000 loops, best of 3: 105 µs per loop
513 - 1000 loops, best of 3: 415 µs per loop
1025 - 1000 loops, best of 3: 1.26 ms per loop
```

scpeters, 2020-05-22:
good call on the memory allocation; I tried copying the implementation of `Image::Pixel(...)` into `Image::MaxColor()` in b8e0a514047e92a60b5a0530561eb3782414acd6, and it speeds it up massively (80 ms for the largest image, down from 764 s). It's a bit heavy-handed, but it seems to do the trick. I'll make a PR

scpeters, 2020-05-22:
see #66

mjcarroll, 2020-06-10:
Is this closed by #66?

PR #66 (2020-06-09): Refactor Image::MaxColor to reduce computation time
This refactors `Image::MaxColor` to massively improve performance for large images, which is a significant bottleneck when loading ImageHeightmaps, as documented in #65. Several all-black images of different sizes (that happen to be suitable for heightmaps) were added in 02ffa90 and a performance test was added to `UNIT_Image_TEST` in d3eb55b. In b8e0a51, I copied the implementation of `Image::Pixel(...)` into `Image::MaxColor` to reduce the allocation and deallocation of FreeImage structs, which @mjcarroll had suggested may be the issue in h [... cut]