gazebosim/gz-common #91: COLLADA Cube Mesh not Colored

jaelrod, 2020-09-01:
Loading a COLLADA mesh containing a cube with colored surface results in a white cube instead.
[mesh.dae.txt](https://github.com/ignitionrobotics/ign-common/files/5157951/mesh.dae.txt)

COLLADA loaded in Blender:
[image]

Same COLLADA loaded in Ignition Gazebo v3.2.0:
[image]

ahcorde, 2021-01-13:
Hi @jaelrod,

I have loaded your mesh into Blender and imported again to Collada and I'm able to see the color:

[image]

A quick look to your dae file looks like your triangles have no material.  You include the material in the triangles:
Replace this line:

```xml
        <triangles count="12">
```
with:

```xml
        <triangles material="material-cube" count="12">
```

Please close the issue if the problem is resolved.

jaelrod, 2021-01-13:
Why would <triangles/> be the place to define the material. If I have a geometries library, I would fundamentally prefer that to be decoupled from the material applied on top of the geometry, which I believe is the inspiration behind the COLLADA specification itself separating the definition of geometry vs. materials. If I want to have a blue cube like this and a cube that is similarly red in hue, I should be able to accomplish that with simply applying a different, red material from library_materials onto the same geometry from library_geometries.

In light of that, I think my application of the material makes more sense where it is in the .dae.txt that I uploaded, in the instantiation of the mesh under the visual_scene, rather than to the triangles themselves:
```
  <library_visual_scenes>
    <visual_scene id="Scene" name="Scene">
      <node>
        <matrix sid="transform">0.35 0 0 0 0 0.4 0 0 0 0 1 0 0 0 0 1</matrix>
        <instance_geometry url="#mesh-cube">
          <bind_material>
            <technique_common>
              <instance_material symbol="material" target="#material-cube"/>
            </technique_common>
          </bind_material>
        </instance_geometry>
      </node>
    </visual_scene>
  </library_visual_scenes>
```

This might seem a bit pedantic, but it does, in fact, make a huge difference for me, as I am composing these assets' information in a rather unconventional way. Since posting this comment, working around this issue has resulted in an order of magnitude increase in size of my COLLADA assets due to repeating equivalent geometries within library_geometries.

As mentioned in the original comment, the *exact* attached mesh file loads into blender with the material correctly applied to the geometry. Per the COLLADA specification, the material reference within the visual_scene ought to be valid.

jaelrod, 2021-01-13:
Khronos Group COLLADA Specification 1.5.0:
https://www.khronos.org/files/collada_spec_1_5.pdf

From **Specification – Core Elements Reference 5-61 (page 101)**:
> The <instance_geometry> element instantiates an object described by a <geometry> element. A
> geometry object is instantiated within the local coordinate system of its parent <node> or <shape> and
> that determines its position, orientation, and scale. COLLADA supports convex mesh, mesh, and spline
> primitives. 
> ...
> [ <bind_material> ] binds material symbols to material instances. This allows a
> single geometry to be instantiated into a scene multiple times
> each with a different appearance. See main entry. 

From **COLLADA – Digital Asset Schema Release 1.5.0 (page 268)**:
> This abstract symbol needs to be bound to a particular material instance. The application does the
> instantiation when processing the <instance_geometry> elements within the <bind_material>
> elements. The application scans the geometry for material attributes and binds actual material objects to
> them as indicated by the <instance_material> (geometry) symbol attributes. See “Example” below.

ahcorde, 2021-01-15:
Thank you for the explanation @jaelrod

I have created a PR that should fix the issue https://github.com/ignitionrobotics/ign-common/pull/151