gazebosim/gz-common #109: Video recordings are not really nice to view

peci1, 2020-10-26:
Is there a specific reason why the default bitrates for video recordings are so low?

https://github.com/ignitionrobotics/ign-common/blob/526c2aceb847dcdbfc2d25fc72457caaece03d3c/av/src/VideoEncoder.cc#L144-L164

[This](https://github.com/ignitionrobotics/ign-common/files/5440500/video.zip) is a 1500x1000 video recorded in current SubT simulator (ign blueprint). If I get it correctly, ign-common uses High profile with slow preset. The result shouldn't be that bad given enough bandwidth.

According to e.g. http://www.lighterra.com/papers/videoencodingh264/ , bitrate for FullHD should be somewhere between 5-8 Mbps to achieve good quality. The paper also states most streaming companies stream FullHD H.264 videos at 3.5-5 Mbps.

Is the reason for choosing such low bandwidth performance?

iche033, 2020-10-26:
yeah the bitrate values could be improved. Those values probably have not changed since it was first written in gazebo several years ago. We noticed the quality issue as well, and instead of changing the default values here (to avoid a behavior change in a release version), we added the ability to configure the bitrate in ign-gazebo via `~/.ignition/gazebo/gui.config`, see ignitionrobotics/ign-gazebo#419

peci1, 2020-10-26:
Thanks, I've stumbled upon this PR. Are there also plans to make this option available in the "start recording" service, so that users can choose the quality during runtime?

iche033, 2020-10-26:
that'll be nice. It's not in our near term plan so I've ticketed ignitionrobotics/ign-gazebo#428

peci1, 2020-10-26:
Thanks. Consider closing this issue if you feel it is "sufficiently resolved" by the ign-gazebo PR. But I'd consider changing the defaults at least in Ignition Gazebo E...