gazebosim/gz-common #112: Top-level Dockerfile for development

adlarkin, 2020-10-29:
_Note: I'm creating this issue here because it relates to all ignition repositories, and I figured that `ign-common` was a good place for generic ignition-related topics. If this issue should be moved elsewhere, let me know!_

As [discussed](https://github.com/ignitionrobotics/ign-gazebo/pull/397#pullrequestreview-514004832) with @chapulina in a [development Docker proposal for `ign-gazebo`](https://github.com/ignitionrobotics/ign-gazebo/pull/397), it would be worth having a top-level Dockerfile that can build the whole ignition collection from source, since users may need source dependencies from multiple repositories when developing/testing new features. I think that there are a few things worth discussing before creating this top-level Dockerfile:

1. If we have a top-level Dockerfile that builds the whole ignition collection from source, which repository should be responsible for maintaining this Dockerfile? Would `ign-common` be a good place, or potentially another repository like [ignition-tooling/gzdev](https://github.com/ignition-tooling/gzdev)?
2. How do we want to handle ignition repository management inside of the container? I think there are three possible options:
    1) Load and build all repositories during image creation
        - **Pros**: Users have access to a container that is already built, allowing for immediate usage of ignition libraries
        - **Cons**: A re-build must occur if users want to use a different branch of a specific repository. This problem can be solved by having users specify which branches they want to use when building the image, but this customization may be difficult to implement. Also, users would have to set up their git account inside of the image if they wanted to make permanent changes to a branch (and if users use SSH for git, this means they'll also have to configure SSH keys as well). Users would also need to configure any development tools they'd like to use inside of the image, unless they're fine with using default configurations of terminal-based editors like `vim`
    2) Load repositories during image creation, but don't build them
        - **Pros**: Users can checkout the specific repository branches they need when a container is started, removing any complexities of trying to specify branch configuration during image build time
        - **Cons**: Users have to build the repositories in the container once their branch configuration has been specified. This can take a while if [ign-physics](https://github.com/ignitionrobotics/ign-physics) needs to be built with tests. Also, just like option 1, git configuration would need to occur inside of the image in order to allow for permanent changes inside of a container (development tool configuration may also need to be set up depending on the user's preferences, as mentioned in option 1)
    3) Install & configure all third-party dependencies during image creation, but leave it up to users to load and build the ignition repositories at container runtime via [volume mounting](https://docs.docker.com/storage/volumes/)
        - **Pros**: Allows for maximum flexibility and minimum image configuration. Users can maintain a permanent copy of the repositories on their machine locally, and test their copy by simply starting a container that has repositories loaded at container run time. This also allows users to make repository changes locally, using development tools that have already been configured. This eliminates the need of setting up anything inside of Docker (Docker is simply used for building and testing)
        - **Cons**: Just like option 2, users would have to build the repositories in the container instead of starting a container with everything pre-built. There can also be a bit of a learning curve with volumes (although from my experience, you learn how volumes work after one or two experiences with them)

More information about pros/cons of the options listed above can be found [here](https://github.com/ignitionrobotics/ign-gazebo/pull/397#issuecomment-713680445).

Does anyone else have thoughts/opinions about the points mentioned above? Also, is there anything else that needs to be considered? (cc @j-rivero)

chapulina, 2020-10-29:
Thank you for bringing this up!

>  which repository should be responsible for maintaining this Dockerfile?

Good question. I'd say `ign-common` is not a good place, despite its name. When it comes to the Ignition stack, this library isn't either a root or a leaf, so it's pretty much there in the middle.

Some of the tooling repositories sound like a good option, but I'd be concerned about unnecessarily exposing a lot of unrelated infrastructure stuff to users.

Some other options:

* Keeping it with the "metapackage" for each collection, i.e.: https://github.com/ignitionrobotics/ign-citadel/ 
* Creating a new repository in the `ignitionrobotics` org for this, like `docker-dev`?

> How do we want to handle ignition repository management inside of the container?

I don't have a clear preference at the moment. My personal development setup is closer to 2.ii, but it's a bit unusual since my containers aren't disposable and I treat them like mini-computers (my Bionic container is more than 1 year old!):

```
CONTAINER ID        IMAGE               COMMAND             CREATED             STATUS                   PORTS               NAMES
b22ab810db2d        devbase-focal       "/bin/bash"         6 months ago        Exited (2) 2 weeks ago                       dev_focal
239178e03f81        devbase-bionic      "/bin/bash"         14 months ago       Up 2 days                                    dev
```

I suspect each developer may have their own preferences like that, especially members of the core team who are constantly jumping across libraries, releases and distros.

To narrow down our target audience, I'd suggest we try to target the casual contributor. Here's a user story:

_As a first-time contributor, I want to build a couple of libraries from source (i.e. `ign-math` and `ign-sensors`), make inter-dependent changes to both libraries, open pull requests to both, and iterate on the reviewer feedback until the pull requests are merged._

What would be the best option for that user?

---

I'll also leave a link to this repository that has a docker setup from some external contributors that are only interested in `ign-math`, maybe there are some ideas to take from there:

https://github.com/ximenesfel/ignition_math_docker

adlarkin, 2020-10-29:
> Some of the tooling repositories sound like a good option, but I'd be concerned about unnecessarily exposing a lot of unrelated infrastructure stuff to users.
>
> Some other options:
>
> * Keeping it with [...]

I do agree that putting this in one of the tooling repositories would expose users to unnecessary/unrelated infrastructure software (and I guess that development tools shouldn't really be considered "infrastructure tools"). While we could put this in the "metapackage" for each collection, this would mean that we'd have to maintain development Docker related files across several repositories, which could be a burden (or maybe it won't be - it's hard to say since we haven't started implementing top-level Dockerfiles yet).

I like the idea of creating a new repository for the following reasons:

1. The files used for Docker development theoretically aren't part of an ignition library - it's just a tool to aid with development. So, placing these files in another existing ignition repository doesn't make sense/could be misleading.
2. I think that we could have just one top-level Dockerfile that would work for all collections if we make use of things like [vcstool](https://github.com/dirk-thomas/vcstool) and a repository's [packages*.apt](https://github.com/ignitionrobotics/ign-gazebo/tree/main/.github/ci) files. If I'm correct in my assumptions of only needing one top-level Dockerfile, then there's no reason to have duplicate Dockerfiles scattered across each collection's "metapackage".

> To narrow down our target audience, I'd suggest we try to target the casual contributor. Here's a user story:
>
> As a first-time contributor, I want to build a couple of libraries from source (i.e. [...]

I'd argue for option ii or iii. With option ii, all the user would need to do (_assuming that git is configured inside of the container_) is create/checkout the proper branches in `ign-math` and `ign-sensors` once the container has been started (assuming that all of the other library branches are correct - which they should be if the repositories were set up in the image using `vcstool`). With option iii, the user can get all of the repositories on their machine locally with `vcstool`, create/checkout the proper branches in `ign-math` and `ign-sensors`, and then pass these repositories into the container when it's time to build/test changes. For both of these options, users could use [colcon](https://colcon.readthedocs.io/en/released/) to build the collection once the proper branches have been selected.

One thing to note though is that with option ii, users would have to re-select their branches if they shut down the container and come back to it later (unless they keep the container running and don't shut it down until the review process is complete). I bring this up because typically, there's a time delay from when a PR is marked ready for review and reviewers give feedback or CI results come in. So, some users may decide to kill the container while they wait for appropriate feedback. This isn't a big deal since switching branches is easy, but it's something users will have to remember if they prefer to start/stop their containers often.

adlarkin, 2020-11-02:
As a follow-up, here are a few great thoughts/recommendations from a few other people I spoke to about [rocker](https://github.com/osrf/rocker):
* Writing a rocker extension for ignition development would mean we wouldn’t have to worry about how to support the user’s graphics card or their weird and unique mouse or anything else. The extension system means we can write our own extensions that get included into the dockerfile as separate python module(s) and distribute them separately using pip or apt packages (or whatever else we prefer)
* @tfoote mentioned the following: "_Also adding in rocker extensions to the core and extending the built in capabilities are also fair game. So far I haven't run into cases where the rocker extensions can't do something that you can do with a generic Dockerfile, and rocker has the ability to do things that a generic Dockefile can't do such as responding to the users' environment such as making sure the UIDs are the same etc and passing through ssh keys or home directory if they are wanted. And you don't have to start to try to have combinatoric support for options parameterized in a "generic" Dockerfile._"

mjcarroll, 2020-11-02:
I spent a little time last week experimenting with a variant of [`setup-ros-docker`](https://github.com/ros-tooling/setup-ros-docker) to create base containers for ignition work.  These are only installing all of the required development dependencies to eventually make CI faster.  The idea is that you should be able to pull eg `ignition-tooling/ubuntu-focal-dome` and get a complete set of dependencies that you could build from source on top of.  The other docker images are (here)[https://github.com/mjcarroll?tab=packages&repo_name=setup-ign-docker]

It does about what you were saying where it uses `vcstool` and the `package.apt` files to install dependencies.  It may be worth a look: https://github.com/mjcarroll/setup-ign-docker/blob/9634cd916a968ac4ea98ff69fae45ae64bdb91f7/setup-ign.sh#L107-L121

zwn, 2020-11-02:
Are the prebuilt docker images publicly available? I am getting this error:
```
$ docker pull docker.pkg.github.com/mjcarroll/setup-ign-docker/setup-ign-docker-ubuntu-bionic-ign-citadel:mr1
Error response from daemon: Get https://docker.pkg.github.com/v2/mjcarroll/setup-ign-docker/setup-ign-docker-ubuntu-bionic-ign-citadel/manifests/mr1: no basic auth credentials
```

j-rivero, 2020-11-02:
Thanks Asthon for ordering ideas and proposals. I think that the idea of @chapulina of having use cases per user profile is a good start (external contributor, core developer, etc.) so we can verify that the final solution is good enough.

One requirements I would put on the table from the top of my head:
 * Handle GPU acceleration (we need to consider at least nvidia but probably we are interested in Intel too) inside the docker container transparently
 * User experience inside that container: we want people to have their configurations loaded
 * Consider the combinations of Linux platforms available: Linux distro, version and architecture
 * Some defaults parameter so new comers don't need to know relationship between ignition releases, platforms and dependencies.

There is a good bunch of useful tools detailed in these comments (or in @mjcarroll's implementation) that are really relevant to implementations. My feeling is that we are going to need an external layer/tool (could be either `rocker` or `gzdev`) to be able to combine requirements together.

> How do we want to handle ignition repository management inside of the container? I think there are three possible options:

My first intention would be to use option II (load all sources, don't build) but docker images could make the option I a bit more interesting and a faster user experience. Note that option III (mount volumes) can suffer from the fact of having different toolchains/dependencies in host system and container.

adlarkin, 2020-11-02:
> Note that option III (mount volumes) can suffer from the fact of having different toolchains/dependencies in host system and container.

@j-rivero can you elaborate on this point a little more? Based on more feedback we have gotten, it sounds like option III would be preferred because it allows users to have a permanent copy of the repositories on their machine (including recent changes), and also avoids having to set up things like editors/IDEs and git settings inside of the Docker container. Just to clarify, if we try to stick to something like option III, the only things that would be mounted as a volume would be the ignition repositories - we'd essentially just have users specify an argument that defines the path to the repositories on their local machine when they are ready to start a container, and then once the container is started, all users would have to do is build the source code from the repositories.

j-rivero, 2020-11-03:
> @j-rivero can you elaborate on this point a little more?

Yes. Sorry I'm probably did not clarify good enough what we try to mount: if that is source code or other non-architecture dependent stuff, that is fine. If when we mount the volume it contains for example a colcon/catkin workspace, where there is a build directory with binaries generated by the host or the container, there is a risk of mixing these binary artificats from compilation which can be potentially incompatible (linked against different versions of dependencies, etc).

adlarkin, 2020-11-03:
> > @j-rivero can you elaborate on this point a little more?
> 
> Yes. Sorry I'm probably did not clarify good enough what we try to mount: if that is source code or other non-platform dependent stuff, that is fine. If when we mount the volume it contains for example a colcon/catkin workspace, where there is a build directory with binaries generated by the host or the container, there is a risk of mixing these binary artificats from compilation which can be potentially incompatible (linked against different versions of dependencies, etc).

I agree - my thought was to create a colcon workspace (or something similar) inside of the Docker image, and then have users load their code into the `src/` directory when the container is started.

If we find that users might want to mount non-source code related things into the container, perhaps this is where something like a rocker extension can be useful.

adlarkin, 2021-01-14:
I'm closing this, now that we have a `ign-docker-env` in https://github.com/ignition-tooling/gzdev.

closed 2021-01-14 by adlarkin