gazebosim/gz-common #121: What is SystemPaths::AddSearchPathSuffix() for?

peci1, 2020-11-04:
https://github.com/ignitionrobotics/ign-common/blob/0d44a6e3d78590207061593c6f503242eea826b6/src/SystemPaths.cc#L505-L518

What is this function for? It sets `this->dataPtr->suffixPaths`, but this field has no usage in ign-common (except for this assignment).

scpeters, 2020-11-04:
I have no idea

chapulina, 2020-11-05:
It looks like it's an incomplete port from `gazebo::common`. There, these paths are inserted between the `path` and `filename` for extra lookups:

https://github.com/osrf/gazebo/blob/063d7386ed5499c2883de48e19ff5fb9dec01dff/gazebo/common/SystemPaths.cc#L426-L438

peci1, 2020-11-05:
Good catch. I'm now working on improving support of common::URI and common::SystemPaths on Windows (as I've found some glitches and now on home-office I have better access to Windows than from university). Should I transplant this piece of code, too? Or should I rather send it as a separate PR?

chapulina, 2020-11-06:
> I'm now working on improving support of common::URI and common::SystemPaths on Windows

That's great, thanks!

>  Should I transplant this piece of code, too? Or should I rather send it as a separate PR?

I'm not sure how large are the changes for Windows. I'd say that if they aren't too big, this could be squeezed in together.