gazebosim/gz-common #122: AV component not being tested on Windows

chapulina, 2020-11-04:
Our CI isn't installing `ffmpeg` on Windows, so the `av` component is being skipped.

Besides being a problem for `ign-common`, the lack of support for AV on Windows means `ign-gazebo` can't be built either.

We started this branch to install the required dependencies:

https://github.com/ignition-tooling/release-tools/compare/master...ign_common

The first run with that branch was taking over 4 hours to build and install `ffmpeg` with vcpkg, so we aborted it:

https://build.osrfoundation.org/job/ignition_common-ci-ign-common3-windows7-amd64/68/

<details><summary>The second run failed with these errors</summary>
<p>

```
Starting package 1/1: ffmpeg:x64-windows
Building package ffmpeg[avresample,core]:x64-windows...
-- Using cached C:/vcpkg/downloads/ffmpeg-ffmpeg-n4.2.tar.gz
-- Using source at C:/vcpkg/buildtrees/ffmpeg/src/n4.2-b7812eeecc
-- Acquiring MSYS Packages...
CMake Error at scripts/cmake/vcpkg_execute_required_process.cmake:72 (message):
    Command failed: C:/vcpkg/downloads/tools/msys2/msys64/usr/bin/bash.exe --noprofile --norc -c "pacman -Sy --noconfirm --needed diffutils make"
    Working Directory: C:/vcpkg/downloads/tools/msys2
    Error code: 1
    See logs for more information:
      C:\vcpkg\buildtrees\ffmpeg\msys-pacman-x64-windows-out.log
      C:\vcpkg\buildtrees\ffmpeg\msys-pacman-x64-windows-err.log

Call Stack (most recent call first):
  scripts/cmake/vcpkg_acquire_msys.cmake:106 (vcpkg_execute_required_process)
  ports/ffmpeg/portfile.cmake:43 (vcpkg_acquire_msys)
  scripts/ports.cmake:90 (include)

Error: Building package ffmpeg:x64-windows failed with: BUILD_FAILED
Please ensure you're using the latest portfiles with `.\vcpkg update`, then
submit an issue at https://github.com/Microsoft/vcpkg/issues including:
  Package: ffmpeg:x64-windows
  Vcpkg version: 2020.02.04-nohash
```

</p>
</details>

j-rivero, 2020-11-05:
Currently blocked by the fact that I'm unable to install ffmpeg from vcpkg due to problems with msys tool used in the installation.

traversaro, 2020-11-06:
> 
> 
> Currently blocked by the fact that I'm unable to install ffmpeg from vcpkg due to problems with msys tool used in the installation.

Which vcpkg tag are you using? There have been a few msys2 related failures in some past version of vcpkg, that should be fixed in the latest vcpkg version. For what regards ffmpeg, there are stil some problems in the .pc files installed by vcpkg, so if you have `pkg-config` installed in your Windows CI this could create failures, see https://github.com/microsoft/vcpkg/pull/11273 for more info. 

On a related note, @Tobias-Fisher recently enabled support for ffmpeg in ignition-common build in conda-forge also on Windows, and at least compilation seems to be working, see https://github.com/conda-forge/libignition-common-feedstock/pull/14 .

j-rivero, 2020-11-06:
> Which vcpkg tag are you using?

We are using `2020-01` "release". But I tried last week latest `master`branch on my local machine and failed. I just update it and things are starting to compile, which is more than I had last week. Thanks @traversaro ! This means that we need to update our vcpkg snapshots, related to https://github.com/ignition-tooling/release-tools/issues/340.

traversaro, 2020-11-06:
Vcpkg devs recently tagged the 2020.11  it could male sense to try it.

chapulina, 2020-12-04:
AV tests are running in Windows Jenkins CI, and they're passing since https://github.com/ignitionrobotics/ign-common/pull/135

closed 2020-12-04 by chapulina