gazebosim/gz-common #140: Add capsule and ellipsoid geometries

chapulina, 2020-12-17:
Now that we have capsule and ellipsoid types on `ign-math` (https://github.com/ignitionrobotics/ign-math/pull/163, https://github.com/ignitionrobotics/ign-math/pull/182) and `sdformat` (https://github.com/osrf/sdformat/pull/389, https://github.com/osrf/sdformat/pull/434), we could also add those types to the `MeshManager`, so that these shapes can be created on `ign-rendering`.

Here's `MeshManager::CreateCylinder` as an example:

https://github.com/ignitionrobotics/ign-common/blob/3027dc0220ff2fbdb8d5d020f98a3799b568afd7/graphics/src/MeshManager.cc#L826-L940

mjcarroll, 2021-02-03:
@chapulina now that #154 and #155 have landed, is there anything else that needs to happen here?

chapulina, 2021-02-04:
I don't think so, closing!

closed 2021-02-04 by chapulina