gazebosim/gz-common #156: 👩‍🌾 Regression of AudioDecoder.CheerFile test on Windows

chapulina, 2021-01-22:
The test is fairly new and has only been run 4 times. It passed the first 3, until 2021/01/14, and failed today, 2021/01/21. See the [test history](https://build.osrfoundation.org/job/ignition_common-ci-ign-common3-windows7-amd64/88/testReport/junit/(root)/AudioDecoder/CheerFile/history/).

Test output:

```
43: [ RUN      ] AudioDecoder.CheerFile
43: [31m[Err] [C:\Jenkins\workspace\ignition_common-ci-ign-common3-windows7-amd64\ws\ign-common\av\src\AudioDecoder.cc:192] [m[31mUnable to open audio file[[m[31m_bad_audio_filename_.wav[m[31m]
43: [m[31m[Err] [C:\Jenkins\workspace\ignition_common-ci-ign-common3-windows7-amd64\ws\ign-common\av\src\AudioDecoder.cc:192] [m[31mUnable to open audio file[[m[31mC:/Jenkins/workspace/ignition_common-ci-ign-common3-windows7-amd64/ws/ign-common/test\data\audio_bad_codec.grf[m[31m]
43: [m[31m[Err] [C:\Jenkins\workspace\ignition_common-ci-ign-common3-windows7-amd64\ws\ign-common\av\src\AudioDecoder.cc:234] [m[31mCouldn't find audio stream.
43: [munknown file: error: SEH exception with code 0xc0000005 thrown in the test body.
43: [  FAILED  ] AudioDecoder.CheerFile (1177 ms)
```

I've seen this SEH exception on other builds before, so it could be some infrastructure issue.

---

https://github.com/osrf/buildfarmer/issues/145

peci1, 2021-02-03:
I ran the test on a physical Windows machine (enabling it) and it ran ok.

chapulina, 2021-02-04:
Thanks for checking, it may be a problem with our infrastructure.