gazebosim/gz-common #161: Depend on ignition-utils1 from v4

chapulina, 2021-01-29:
In order to use `ign-utils` in `ign-common`, we need to add it as a dependency on all our CI and release infrastructure:

- [ ] `gazebodistro`
- [ ] `release-tools`
- [ ] `homebrew-simulation`
- [ ] `ignition-common4-release`

Besides in the source code, which is done in https://github.com/ignitionrobotics/ign-common/pull/160

See the pull requests linked to https://github.com/ignitionrobotics/ign-gazebo/issues/568 for examples.

scpeters, 2021-01-29:
adding the homebrew dependency in https://github.com/osrf/homebrew-simulation/pull/1313

scpeters, 2021-03-01:
updating the debian package name to libignition-utils1-dev in https://github.com/ignition-release/ign-common4-release/pull/4

scpeters, 2021-03-01:
I think we should use `packages.apt` instead of the change in https://github.com/ignition-tooling/release-tools/pull/391

scpeters, 2021-03-01:
> I think we should use `packages.apt` instead of the change in [ignition-tooling/release-tools#391](https://github.com/ignition-tooling/release-tools/pull/391)

added 640a00d in #160