gazebosim/gz-common #167: ign_remotery_vis script prevents side-by-side installation of common3, common4

scpeters, 2021-02-05:
The `ign_remotery_vis` script is installed to `<prefix>/bin` with no major version name, which causes a conflict between `ign-common3` and `ign-common4`. I expect this will be a problem with side-by-side install of the debs, but I first noticed it with homebrew bottles in https://github.com/osrf/homebrew-simulation/pull/1324:

* [image] https://build.osrfoundation.org/job/generic-release-homebrew_triggered_bottle_builder/259/

~~~
==> Pouring ignition-common4-3.999.999~0~20210125~260e23.mojave.bottle.tar.gz
The formula built, but is not symlinked into /usr/local
Error: The `brew link` step did not complete successfully
Could not symlink bin/ign_remotery_vis
Target /usr/local/bin/ign_remotery_vis
is a symlink belonging to ignition-common3.
~~~

chapulina, 2021-02-05:
We could append the version number to the executable starting from v4.

scpeters, 2021-02-06:
should it be part of an `ign` command? `ign remotery --vis`?

chapulina, 2021-02-06:
Yeah that's an option. It adds a dependency, but I think it's worth it. I'd suggest something like `ign profiler remotery --vis` though, in case we support other profilers in the future.

mjcarroll, 2021-02-08:
Going to use this as an opportunity to fix #82 as well.

In terms of using it with `ign`, we can probably install/register it in such a way that there isn't a direct dependency on ign-tools, but will work if ign-tools is installed.

chapulina, 2021-03-01:
I believe this has been fixed

closed 2021-03-01 by chapulina