gazebosim/gz-common #173: Support loading meshes in glTF format

mahiuchun, 2021-02-12:
glTF 2.0 is a new standard 3D mesh format that supports PBR natively.

## Desired behavior
`MeshManager::Load()` be able to load glTF 2.0 mesh files with file extensions `.glb` and `.gltf`.

## Alternatives considered
N/A

## Implementation suggestion
We can use https://github.com/syoyo/tinygltf to parse glTF files.

## Additional context
https://www.khronos.org/gltf/

guyVaarst, 2024-06-04:
I am a bit confused - this ticket implies that gazebo doesnt yet support gltf, but https://gazebosim.org/docs/all/release-features  implies it does. Does Gazebo support glTF?

azeey, 2024-06-04:
Yes, glTF support was added in https://github.com/gazebosim/gz-common/pull/393. I'll close this issue.

closed 2024-06-04 by azeey