gazebosim/gz-common #181: Build failure on Fedora 34

richmattes, 2021-03-09:
## Environment
* OS Version: Fedora 34 (pre-release)
* Source or binary build? Source - ignition-common-3.10.1
  
  

## Description
When building ignition-common on Fedora 34 and higher, I am experiencing a build failure when compiling `plugin.cc` for the `INTEGRATION_plugin` test.  The command line (see Output section) doesn't include the necessary `-std=c++17`, instead it includes `-std=c++11`.

There are a couple of conflicting C++ standards being set in different places:
- `libignition_common3` has an `INTERFACE` `target_compile_features` set for `cxx_std_17`
- `test/CMakeLists.txt` appends `std=c++11` to the `CMAKE_CXX_FLAGS` before building gtest targets and including test subdirectories.  

Given that the build works as expected on Fedora 33 with CMake 3.18, my guess is something changed in CMake 3.19 where the `CMAKE_CXX_FLAGS` take precedence and isn't overridden by target properties.

The `CMAKE_CXX_FLAGS` in `test/CMakeLists.txt` should be removed in favor of target-specific standards version selection if needed for the gtest libraries. 

* Expected behavior: 
Package builds as expected, including all tests

* Actual behavior: 
Building the INTEGRATION_plugin test fails with an error (C++17 required for `string_view` - see the Output section) 

## Steps to reproduce

1. Download CMake 3.19 or higher
2. Configure and build with the CMake 3.19 binary

## Output
Successful, and failed logs are (here)[https://copr.fedorainfracloud.org/coprs/g/robotics-sig/gazebo-latest/build/2059613/] in the builder_live logs.  Excerpt of a failure case is shown below

```
[ 56%] Building CXX object test/integration/CMakeFiles/INTEGRATION_plugin.dir/plugin.cc.o
cd /builddir/build/BUILD/ignition-common-3.10.1/aarch64-redhat-linux-gnu/test/integration && /usr/bin/g++  -I/builddir/build/BUILD/ignition-common-3.10.1/include -I/builddir/build/BUILD/ignition-common-3.10.1/aarch64-redhat-linux-gnu/include -I/builddir/build/BUILD/ignition-common-3.10.1/test/gtest/include -I/builddir/build/BUILD/ignition-common-3.10.1/test/gtest -I/builddir/build/BUILD/ignition-common-3.10.1/test -I/builddir/build/BUILD/ignition-common-3.10.1/aarch64-redhat-linux-gnu -I/builddir/build/BUILD/ignition-common-3.10.1 -I/builddir/build/BUILD/ignition-common-3.10.1/aarch64-redhat-linux-gnu/core/include -isystem /usr/include/uuid -O2 -flto=auto -ffat-lto-objects -fexceptions -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1  -mbranch-protection=standard -fasynchronous-unwind-tables -fstack-clash-protection -std=c++11 -DNDEBUG  -Wall -Wextra -Wno-long-long -Wno-unused-value -Wfloat-equal -Wshadow -Winit-self -Wswitch-default -Wmissing-include-dirs -pedantic  -I/usr/include/uuid -o CMakeFiles/INTEGRATION_plugin.dir/plugin.cc.o -c /builddir/build/BUILD/ignition-common-3.10.1/test/integration/plugin.cc
In file included from /builddir/build/BUILD/ignition-common-3.10.1/include/ignition/common/Console.hh:25,
                 from /builddir/build/BUILD/ignition-common-3.10.1/test/integration/plugin.cc:28:
/builddir/build/BUILD/ignition-common-3.10.1/include/ignition/common/Util.hh:197:60: warning: 'visibility' attribute ignored [-Wattributes]
  197 |     constexpr uint64_t IGNITION_COMMON_VISIBLE hash64(std::string_view _key)
      |                                                            ^~~~~~~~~~~
/builddir/build/BUILD/ignition-common-3.10.1/include/ignition/common/Util.hh:197:60: error: 'string_view' is not a member of 'std'
/builddir/build/BUILD/ignition-common-3.10.1/include/ignition/common/Util.hh:197:60: note: 'std::string_view' is only available from C++17 onwards
gmake[2]: *** [test/integration/CMakeFiles/INTEGRATION_plugin.dir/build.make:85: test/integration/CMakeFiles/INTEGRATION_plugin.dir/plugin.cc.o] Error 1
gmake[2]: Leaving directory '/builddir/build/BUILD/ignition-common-3.10.1/aarch64-redhat-linux-gnu'
gmake[1]: *** [CMakeFiles/Makefile2:1984: test/integration/CMakeFiles/INTEGRATION_plugin.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
```

richmattes, 2021-03-09:
I've confirmed that changing `c++11` to `c++17`, and removing the modification of `CMAKE_CXX_FLAGS` in `test/CMakeLists.txt` altogether both resolve the failure.

mjcarroll, 2021-05-06:
@richmattes Can you check #214 for me?

mjcarroll, 2021-05-14:
Closed by #214

PR #214 (2021-05-14): Remove CMAKE_CXX_FLAGS from test targets
# 🦟 Bug fix

Fixes #181 

## Summary

In general, we should never be setting or appending `CMAKE_CXX_FLAGS`, as these can be controlled via target specific flags.

## Checklist
- [x] Signed all commits for DCO
- [x] `codecheck` passed (See [contributing](https://ignitionrobotics.org/docs/all/contributing#contributing-code))
- [x] All tests passed (See [test coverage](https://ignitionrobotics.org/docs/all/contributing#test-coverage))
- [x] While waiting for a review on your PR, please help review [another open pull request](https://github.com/pulls?q=is% [... cut]
commit 32fd74e3 (2021-05-14): Remove CMAKE_CXX_FLAGS from test targets (#214)

* Remove CMAKE_CXX_FLAGS from test targets
* Set target versions for test plugin libs
* Add <thread> to Util

Fixes #181
closed 2021-05-14 by mjcarroll