gazebosim/gz-common #202: Collada Exporter not taking into account all the material/effects

Blast545, 2021-04-14:
## Description
* Expected behavior: 

When using the Collada Exporter from ign-gazebo, all textures files should be exported into the final result.

* Actual behavior: 
They only get exported if their full path name includes "meshes/". 

## Steps to reproduce
In an environment that includes an Ignition installation:

1. Add the ColladaExporter plugin to the `tunnel.sdf` ign-gazebo example file.
2. Run the Collada Exporter on that file.
ign gazebo -v 3 -s -r --iterations 1 tunnel.sdf
3. Open the .dae exported to the path ./tunnel/meshes/tunnel.sdf

However, this can't be reproduced at the moment because it depends on fixing https://github.com/ignitionrobotics/ign-gazebo/issues/761.

I was able to track this problem to these conditional include statements like this one:
https://github.com/ignitionrobotics/ign-common/blob/d496d79d9d90b9474e7ccd298210115084e472ad/graphics/src/ColladaExporter.cc#L511

The collada exporter is only taking into account effects/textures if the path included "meshes/' in its name. However, `ign-gazebo` is able to open all files in its environment path variable and fuel models are usually loaded from `~/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/`, in neither of those cases the file includes the `meshes`.

I imagine there was an old context for this check statement, I didn't find any in the history of this file though. If there are no problems I can open a PR to remove those checks or add to the check at least paths including "materials" as well. Let me know your thoughts.

Blast545, 2021-04-14:
WDYT @gonzodepedro

chapulina, 2021-04-14:
Ignition shouldn't enforce any specific paths, so I think it's ok to remove that check. Let's make sure we add a test exporting a texture without `meshes` in the path to check it works as intended.

---

I can't remember why it was done like that. You can see the commit history on Gazebo classic (the files were ported from there to Ignition):

https://github.com/osrf/gazebo/blame/gazebo11/gazebo/common/ColladaExporter.cc#L359

I didn't find any specific discussion about this on the [pull request backup](https://osrf-migration.github.io/gazebo-gh-pages/#!/osrf/gazebo/pull-requests/1064/page/1).

I originally wrote the `ColladaExporter` to simplify meshes for GzWeb, and GzWeb does enforce folder structures like `meshes` and `materials/textures` for assets, mainly because it's tough to traverse "directories" on the web. So my guess is that the reason is related to that, and it shouldn't be an issue here.

PR #204 (2021-04-26): Remove ColladaExporter path constraint

# 🦟 Bug fix

Fixes #202 

## Summary
This PR removes the ColladaExporter constraint on the path to include "meshes" in their names to be exported. As per discussion in #202 this check might be related to code from Ignition Gazebo and may not be needed now.

## Checklist
- [x] Signed all commits for DCO
- [x] Added tests
- [ ] Updated documentation (as needed)
- [ ] Updated migration guide (as needed)
- [x] `codecheck` passed (See [contributing](https://ignitionrobotics.org/docs/all/contributing#contributing-code))
- [x] All tests passed (See [test covera [... cut]
closed 2021-04-26 by Blast545