gazebosim/gz-common #203: New AV warnings on Homebrew

chapulina, 2021-04-14:
## Environment
* OS Version: macOS Mojave
* Source or binary build? Source, version 3
  
  

## Description
* Expected behavior: No warnings 
* Actual behavior: 4 warnings, turning the build yellow 

There are [4 new deprecation warnings](https://build.osrfoundation.org/job/ignition_common-ci-ign-common3-homebrew-amd64/104/) on macOS. This is most likely caused by an update of an upstream dependency. Due to Homebrew's rolling nature, we unfortunately need to be addressing these as they come. It would be great to have some stability on the macOS side like we have on Linux.

## Steps to reproduce

Run Homebrew CI for this repository.

## Output

```
/Users/jenkins/workspace/ignition_common-ci-pr_any-homebrew-amd64/ign-common/av/src/AudioDecoder.cc:113:3: warning: 'av_init_packet' is deprecated [-Wdeprecated-declarations]
  av_init_packet(&packet);
  ^
/usr/local/Cellar/ffmpeg/4.4/include/libavcodec/packet.h:487:1: note: 'av_init_packet' has been explicitly marked deprecated here
attribute_deprecated
^
/usr/local/Cellar/ffmpeg/4.4/include/libavutil/attributes.h:100:49: note: expanded from macro 'attribute_deprecated'
#    define attribute_deprecated __attribute__((deprecated))
                                                ^
```

```
/Users/jenkins/workspace/ignition_common-ci-pr_any-homebrew-amd64/ign-common/av/src/VideoEncoder.cc:801:5: warning: 'av_init_packet' is deprecated [-Wdeprecated-declarations]
    av_init_packet(avPacket);
    ^
/usr/local/Cellar/ffmpeg/4.4/include/libavcodec/packet.h:487:1: note: 'av_init_packet' has been explicitly marked deprecated here
attribute_deprecated
^
/usr/local/Cellar/ffmpeg/4.4/include/libavutil/attributes.h:100:49: note: expanded from macro 'attribute_deprecated'
#    define attribute_deprecated __attribute__((deprecated))
                                                ^
/Users/jenkins/workspace/ignition_common-ci-pr_any-homebrew-amd64/ign-common/av/src/VideoEncoder.cc:895:7: warning: 'av_init_packet' is deprecated [-Wdeprecated-declarations]
      av_init_packet(avPacket);
      ^
/usr/local/Cellar/ffmpeg/4.4/include/libavcodec/packet.h:487:1: note: 'av_init_packet' has been explicitly marked deprecated here
attribute_deprecated
^
/usr/local/Cellar/ffmpeg/4.4/include/libavutil/attributes.h:100:49: note: expanded from macro 'attribute_deprecated'
#    define attribute_deprecated __attribute__((deprecated))
                                                ^
```

CC @scpeters

scpeters, 2021-04-19:
this was deprecated in the following patch:

* https://www.mail-archive.com/ffmpeg-devel@ffmpeg.org/msg115189.html

the patch has some clues for migration

scpeters, 2021-04-19:
started work in https://github.com/ignitionrobotics/ign-common/commit/d6b78444ea852e73aa418c1507e58f9dd3f4acae

chapulina, 2021-04-22:
Do we need ifdefs to support different ffmpeg versions?

scpeters, 2021-04-22:
> Do we need ifdefs to support different ffmpeg versions?

I used existing ifdefs for my first approach in d6b7844; I think the rest of the cases will need new ifdefs

chapulina, 2021-04-22:
Got it, let me know if I can help as buildfarmer :woman_farmer:

mjcarroll, 2021-05-14:
Fixed by #220

PR #220 (2021-05-14): Fix av_* API usage for deprecations
# 🦟 Bug fix

Fixes #203

## Summary

After this macOS and other platforms with newer ffmpeg/libav version (Fedora 34) should build without deprecation warnings.

## Checklist
- [x] Signed all commits for DCO
- [ ] Added tests
- [ ] Updated documentation (as needed)
- [ ] Updated migration guide (as needed)
- [ ] `codecheck` passed (See [contributing](https://ignitionrobotics.org/docs/all/contributing#contributing-code))
- [ ] All tests passed (See [test coverage](https://ignitionrobotics.org/docs/all/contributing#test-coverage))
- [ ] While waiting for a rev [... cut]
closed 2021-05-14 by mjcarroll