gazebosim/gz-common #239: igniton-common3 uses string_view with c++11 which is a feature of c++17

j-rivero, 2021-08-18:
## Environment
* OS Version: Debian Sid, GCC11
* ign-common3 in debian, version 3.5.0

## Description
* Expected behavior: compilation to be fine with any compiler
* Actual behavior: reports of failing on gcc11 in Debian https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=984182

## Steps to reproduce

1. Install gcc11
2. Compile code

Not sure how we want to resolve the problem. Seems like the own gtest is using `absl::string_view` which maybe an alternative until we have ign-common with c++17 enabled.

scpeters, 2021-08-19:
it's been in Util.hh since 3.0.0:

https://github.com/ignitionrobotics/ign-common/blob/ignition-common3_3.0.0/include/ignition/common/Util.hh#L160

traversaro, 2021-08-22:
> Seems like the own gtest is using `absl::string_view` which maybe an alternative until we have ign-common with c++17 enabled.

I don't know about `absl::string_view` in particular, but in general abseil data structures are quite a nightmare from the ABI point of view (see https://abseil.io/about/compatibility). In general abseil do not promise any kind of ABI compatibility, and even if you fix the abseil version, the ABI of the data structure can change from compilation unit to compilation unit depending on the C++ standard version used to compile that specific compilation unit (so you could get a point in which you have ignition-common compiled with c++11 and a downstream project that uses ignition-common compiled with C++17, and get runtime crashes due to the ABI incompatibilities).