gazebosim/gz-common #245: Potential bug in Collada exporter

caguero, 2021-09-07:
## Environment
* OS Version: Ubuntu 18.04 and Ubuntu 20.04
* Source or binary build? Source, compiled from main (f48085bc5d72b312f4ed0d1721a13565d4f13eb7)
* If this is a GUI or sensor rendering bug, describe your GPU and rendering system. Otherwise delete this section. 
    - Rendering plugin: [ogre | ogre2]. 
      - [ ] Sensor rendering error. 
      - [ ] GUI rendering error. 
    - EGL headless mode:
      - [ ] Running in EGL headless mode 
    - Generally, mention all circumstances that might affect rendering capabilities: 
      - [ ] running on a dual GPU machine (integrated GPU + discrete GPU)
      - [ ] running on a multi-GPU machine (it has multiple discrete GPUs)
      - [ ] running on real hardware
      - [ ] running in virtual machine
      - [ ] running in Docker/Singularity
      - [ ] running remotely (e.g. via SSH)
      - [ ] running in a cloud
      - [ ] using VirtualGL, XVFB, Xdummy, XVNC or other indirect rendering utilities
      - [ ] GPU is concurrently used for other tasks
        - [ ] desktop acceleration
        - [ ] video decoding (i.e. a playing Youtube video)
        - [ ] video encoding
        - [ ] CUDA/ROCm computations (Tensorflow, Torch, Caffe running)
        - [ ] multiple simulators running at the same time
      - [ ] other...
    - Rendering system info:
      - On Linux, provide the outputs of the following commands:
          ```bash
          LANG=C lspci -nn | grep VGA  # might require installing pciutils
          echo "$DISPLAY"
          LANG=C glxinfo -B | grep -i '\(direct rendering\|opengl\|profile\)'  # might require installing mesa-utils package
          ps aux | grep Xorg
          sudo env LANG=C X -version  # if you don't have root access, try to tell the version of Xorg e.g. via package manager
          ```
      - On Windows, run `dxdiag` and report the GPU-related information.
      - On Mac OS, open a terminal and type `system_profiler SPDisplaysDataType`. Copy the output here.
        
    - [ ] Please, attach the ogre.log or ogre2.log file from  `~/.ignition/rendering` 
    
I have two almost exact `.dae` meshes. One of them loads and the other doesn't with the following error message:

```
[GUI] [Err] [Ogre2MeshFactory.cc:559] Cannot load mesh with zero sub-meshes
[GUI] [Err] [Ogre2MeshFactory.cc:125] Failed to get Ogre item for [file:///home/caguero/vrx2022_ws/src/vrx/vrx_gazebo/models/placard_2022/meshes/Placard.dae]
[GUI] [Err] [SceneManager.cc:382] Failed to load geometry for visual: visual
```

[placard.zip](https://github.com/ignitionrobotics/ign-common/files/7122820/placard.zip)
  
See the attached `.zip` containing two meshes. `placard_2022.dae` loads but `Placard.dae` doesn't load. I recommend to run `meld` to see the differences. The diff shows just a difference in `scene id` name and `<instance_visual_scene url=`.

## Steps to reproduce

1. Drag and drop `placard_2022.dae` into the scene. The  mesh should load.
2. Drag and drop `Placard.dae` into the scene. The  mesh doesn't load.

mjcarroll, 2021-09-07:
> Source or binary build? Source, compiled from main (f48085bc5d72b312f4ed0d1721a13565d4f13eb7)

Are you compiling from `main` or `ign-common4`?  I'm having trouble locating this specific commit on either branch:

```
git show f48085bc --stat
fatal: ambiguous argument 'f48085bc': unknown revision or path not in the working tree.
```

mjcarroll, 2021-09-07:
I just did a quick test with the `ColladaLoader` class and the difference seems to be: 

```
 RUN      ] ColladaLoader.LoadPlacard
[       OK ] ColladaLoader.LoadPlacard (0 ms)
[ RUN      ] ColladaLoader.LoadPlacard_2022
[Err] [SystemPaths.cc:467] Could not resolve file [Placard.png]
[Err] [Material.cc:164] Unable to find texture [Placard.png] as a locally cached texture or in path [/home/mjcarroll/workspaces/ign_fortress/src/ign-common/test/data]
[       OK ] ColladaLoader.LoadPlacard_2022 (1 ms)
```

Is there an image that is missing here?

caguero, 2021-09-07:
> Are you compiling from `main` or `ign-common4`? I'm having trouble locating this specific commit on either branch:
>
> ```
> git show f48085bc --stat
> fatal: ambiguous argument 'f48085bc': unknown [...]

Sorry, wrong commit. The good one is e70a75d107ba86bb3d1a68690e1d33bfaa4cbb29 . I'm using `ign-common4`.

caguero, 2021-09-07:
> I just did a quick test with the `ColladaLoader` class and the difference seems to be:
>
> ```
>  RUN      ] ColladaLoader.LoadPlacard
> [       OK ] ColladaLoader.LoadPlacard (0 ms)
> [ RUN      ] [...]

Not sure if it's related because one of the meshes loads but just in case I have included in the `zip` file.

mjcarroll, 2021-09-07:
Added these models as a test here: https://github.com/ignitionrobotics/ign-common/tree/placard_test

Can confirm that it fails:

```
[ RUN      ] ColladaLoader.LoadPlacard
/home/mjcarroll/workspaces/ign_fortress/src/ign-common/graphics/src/ColladaLoader_TEST.cc:499: Failure
Expected equality of these values:
  6u
    Which is: 6
  mesh->VertexCount()
    Which is: 0
/home/mjcarroll/workspaces/ign_fortress/src/ign-common/graphics/src/ColladaLoader_TEST.cc:500: Failure
Expected equality of these values:
  6u
    Which is: 6
  mesh->NormalCount()
    Which is: 0
/home/mjcarroll/workspaces/ign_fortress/src/ign-common/graphics/src/ColladaLoader_TEST.cc:501: Failure
Expected equality of these values:
  6u
    Which is: 6
  mesh->IndexCount()
    Which is: 0
/home/mjcarroll/workspaces/ign_fortress/src/ign-common/graphics/src/ColladaLoader_TEST.cc:502: Failure
Expected equality of these values:
  6u
    Which is: 6
  mesh->TexCoordCount()
    Which is: 0
/home/mjcarroll/workspaces/ign_fortress/src/ign-common/graphics/src/ColladaLoader_TEST.cc:503: Failure
Expected equality of these values:
  2u
    Which is: 2
  mesh->SubMeshCount()
    Which is: 0
[  FAILED  ] ColladaLoader.LoadPlacard (0 ms)
[ RUN      ] ColladaLoader.LoadPlacard_2022
[       OK ] ColladaLoader.LoadPlacard_2022 (1 ms)
```

And the diff is as you say:

```
9c9
<     <created>2021-08-31T22:23:17Z</created>
---
>     <created>2021-08-31T21:25:29Z</created>
11c11
<     <modified>2021-08-31T22:23:17Z</modified>
---
>     <modified>2021-08-31T21:25:29Z</modified>
577c577
<     <visual_scene id="placard_2022" name="placard_2022">
---
>     <visual_scene id="Placard" name="Placard">
605c605
<     <instance_visual_scene url="#placard_2022"/>
---
>     <instance_visual_scene url="#Placard"/>
```

fmrico, 2024-10-06:
Hi, 

I can confirm the problem persists in Gazebo Sim, version 8.6.0 (binary package for Jazzy)