gazebosim/gz-common #317: Inconsistent signed/unsigned integer type in SubMesh MaterialIndex

scpeters, 2022-03-01:
As noted in https://github.com/ignitionrobotics/sdformat/pull/831/files#r816364549, the [SubMesh::MaterialIndex()](https://github.com/ignitionrobotics/ign-common/blob/ignition-common4_4.5.0/graphics/include/ignition/common/SubMesh.hh#L287-L289) accessor method and corresponding setter method both use `unsigned int` for the material index, but the value is stored internally as an `int` in [Material.cc](https://github.com/ignitionrobotics/ign-common/blob/ignition-common4_4.5.0/graphics/src/SubMesh.cc#L55) and explicitly uses a value of `-1` to represent an unsigned value.

We should consider using `std::optional<unsigned int>` for this API in ign-common5.

chapulina, 2022-03-01:
> We should consider using std::optional<unsigned int> for this API in ign-common5.

That would be an elegant solution. A less elegant one, but that doesn't require a deprecation cycle and maintains API, would be to return something like `std::numeric_limits<unsigned int>::max()` for an invalid material.

adlarkin, 2022-03-01:
I have a PR in progress that takes the `std::numeric_limits<unsigned int>::max()` approach for `ign-common5`. Should I continue down that path, or should I update the PR to use `std::optional`? I think that `std::optional` is the better option in this scenario and would prefer that approach, even if it requires tick tock.

chapulina, 2022-03-01:
I'm ok with optional :+1: tick-tocking return values can be tricky though, we may need to change the getter's name

adlarkin, 2022-03-03:
I opened a PR that uses `std::optional` in a new method and deprecates the current `MaterialIndex` method: #319

mjcarroll, 2022-04-18:
Closed by #319

PR #319 (2022-04-05): Fix edge case handling for SubMesh::MaterialIndex

# 🦟 Bug fix

Fixes #317

## Summary
As mentioned in #317, `SubMesh::MaterialIndex` returns an `unsigned int`, but internally, this data is stored in an `int` with an initial value of `-1` to represent a submesh having no material index applied to it.

To avoid implicit casts, I've created a new method named `GetMaterialIndex` that replaces the current `MaterialIndex` method. This new method returns a `std::optional<unsigned int>` to indicate whether the submesh has a material index assigned to it or not. The current `MaterialIndex` method is n [... cut]
closed 2022-04-18 by mjcarroll