gazebosim/gz-common #318: Build error when installing from archlinux AUR

capstayn, 2022-03-02:
## Environment
* OS Version: archlinux, up to date
* Source: the AUR page: https://aur.archlinux.org/ignition-common.git, version 4.5.0-2

Installation with 

`$ yay -S ignition-common
`
fails, with this log: 
    
```
/home/pedro/.cache/yay/ignition-common/src/ign-common-ignition-common4_4.5.0/av/src/AudioDecoder.cc: In member function ‘bool ignition::common::AudioDecoder::Decode(uint8_t**, unsigned int*)’:
/home/pedro/.cache/yay/ignition-common/src/ign-common-ignition-common4_4.5.0/av/src/AudioDecoder.cc:135:24: error: ‘avcodec_decode_audio4’ was not declared in this scope; did you mean ‘avcodec_decode_subtitle2’?
  135 |         bytesDecoded = avcodec_decode_audio4(this->data->codecCtx, decodedFrame,
      |                        ^~~~~~~~~~~~~~~~~~~~~
      |                        avcodec_decode_subtitle2
/home/pedro/.cache/yay/ignition-common/src/ign-common-ignition-common4_4.5.0/av/src/AudioDecoder.cc: In member function ‘bool ignition::common::AudioDecoder::SetFile(const string&)’:
/home/pedro/.cache/yay/ignition-common/src/ign-common-ignition-common4_4.5.0/av/src/AudioDecoder.cc:227:44: error: ‘AVStream’ {aka ‘struct AVStream’} has no member named ‘codec’
  227 |     if (this->data->formatCtx->streams[i]->codec->codec_type == // NOLINT(*)
      |                                            ^~~~~
/home/pedro/.cache/yay/ignition-common/src/ign-common-ignition-common4_4.5.0/av/src/AudioDecoder.cc:253:31: error: ‘AVStream’ {aka ‘struct AVStream’} has no member named ‘codec’
  253 |     this->data->audioStream]->codec;
      |                               ^~~~~
/home/pedro/.cache/yay/ignition-common/src/ign-common-ignition-common4_4.5.0/av/src/AudioDecoder.cc:259:43: error: invalid conversion from ‘const AVCodec*’ to ‘AVCodec*’ [-fpermissive]
  259 |   this->data->codec = avcodec_find_decoder(this->data->codecCtx->codec_id);
      |                       ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                           |
      |                                           const AVCodec*
make[2]: *** [av/src/CMakeFiles/ignition-common4-av.dir/build.make:76: av/src/CMakeFiles/ignition-common4-av.dir/AudioDecoder.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:395: av/src/CMakeFiles/ignition-common4-av.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
==> ERROR: A failure occurred in build().
    Aborting... 
error making: ignition-common
```

Maybe `ignition-common` needs some different versions of the packages installed on the system?

CoLuthien, 2022-03-10:
https://github.com/FFmpeg/FFmpeg/blob/release/5.0/libavcodec/decode.c

avcodec_decode_audio4 removed at v5.0. 

Though I won't find removed commit :( 
i'll try with v4.4 ffmpeg :)

mjcarroll, 2022-03-10:
We currently don't have support for FFmpeg 5.0+.

For now, please use an older version, and I'm going to close this as a duplicate of #306, where we are tracking the same thing (homebrew on macos is also a rolling package manager)

closed 2022-03-10 by mjcarroll