gazebosim/gz-common #330: Compile error in geospatial/src/Dem.cc

darksylinc, 2022-03-22:
## Environment
* OS Version: Ubuntu 18.04
* Source or binary build?
  Source, main. Commit ff5970e7aebb102eb38aa7d780bbdb015ea6544d

Trying to run latest ign-rendering7 fails because ign-common5's CMake tries to pull _something_-geospatial.cmake

But that won't exist unless GDAL dependency is installed.

So I installed GDAL, rebuilt from scratch and met with this gem:

```
--- stderr: ignition-common5                                                                                   
/home/matias/Projects/ign/src/ign-common/geospatial/src/Dem.cc: In member function ‘bool ignition::common::Dem::GeoReference(double, double, ignition::math::v7::Angle&, ignition::math::v7::Angle&) const’:
/home/matias/Projects/ign/src/ign-common/geospatial/src/Dem.cc:264:28: error: invalid conversion from ‘const char**’ to ‘char**’ [-fpermissive]
     sourceCs.importFromWkt(&importString);
                            ^~~~~~~~~~~~~
In file included from /home/matias/Projects/ign/src/ign-common/geospatial/src/Dem.cc:21:
/usr/include/gdal/ogr_spatialref.h:185:32: note:   initializing argument 1 of ‘OGRErr OGRSpatialReference::importFromWkt(char**)’
     OGRErr      importFromWkt( char ** );
                                ^~~~~~~
make[2]: *** [geospatial/src/CMakeFiles/ignition-common5-geospatial.dir/Dem.cc.o] Error 1
make[1]: *** [geospatial/src/CMakeFiles/ignition-common5-geospatial.dir/all] Error 2
```

## Description
* Expected behavior: To compile correctly (or alternatively why is ign-common5 marked as NOTFOUND if geospatial is skipped?)
* Actual behavior: Compiler error (or alternatively, ign-common5 is being marked as NOTFOUND if GDAL is missing, despite successfully compiling)

## Steps to reproduce

1. Pull latest main.
2. `colcon build --cmake-args -DBUILD_TESTING=OFF -DCMAKE_BUILD_TYPE=Debug -DBUILD_DOCS=OFF --merge-install` **NOTE:** Debug is the build type.

scpeters, 2022-03-22:
the `main` branch of ign-common (major version 5) will be part of the Garden collection, which will be supported on Ubuntu 20.04 and 22.04:

* https://ignitionrobotics.org/docs/garden/install#supported-platforms

so the failures on 18.04 are expected. Perhaps we should improve the messaging about which Ubuntu versions are supported?

chapulina, 2022-03-22:
What @scpeters said is right. Let me know how we can make this clearer.

We've announced the supported platforms on [this post](https://community.gazebosim.org/t/ignitions-supported-platforms/1210), which links to this issue:

* https://github.com/ignition-tooling/release-tools/issues/597

We've also explicitly disabled Bionic CI on the `main` branches, i.e.

* https://github.com/ignitionrobotics/ign-rendering/pull/583

I'll close this issue, but please reopen if the problem comes up in a supported platform.

closed 2022-03-22 by chapulina