gazebosim/gz-common #349: TODO: Deprecate Plugin from Garden

methylDragon, 2022-05-03:
From: https://github.com/ignitionrobotics/ign-common/pull/348#discussion_r863354402

The `Plugin` files are deprecated from Garden, so we should be removing them.

chapulina, 2022-05-03:
I started adding deprecation warnings to plugins a while back and lost that branch 🙃 I'll go ahead and do that now.

> so we should be removing them.

This is referring to the `gz/*Plugin*.hh` header files. Fortress (`common4`) users should migrate from common plugins  (`ignition/*Plugin*.hh`) to the [gz-plugin](https://github.com/ignitionrobotics/ign-plugin/) library (`gz/plugin/*.hh`), so we don't need to provide new `gz/*Pluign*.hh` headers.

---

Note that deprecating (tick-tocking) the plugins is an alternative to the proposal in #336 , which I believe we decided against.

chapulina, 2022-05-03:
* See https://github.com/ignitionrobotics/ign-common/pull/350

PR #350 (2022-05-26): Deprecate plugin in favor of gz-plugin
* Closes #349 

## Summary

* 0e4a0f5de63f47daf8170404a5a3fdc7802353ec: Move plugin headers back to `ignition` folder because users shouldn't migrate to `gz/common/*Plugin*`, but migrate to `gz/plugin` instead.
* 7090764daa6afefd4812fe39aafaf32e317e57be: Use the correct headers
* e0f0e216a48082d310ef30ef86df793813fbb67d: Add deprecation warnings

## Test it

CI should pass without warnings.

My local cppcheck had some `unknownMacro` warnings, but that may be due to my version. I'll see what CI says and suppress those warnings as needed.

## Checklist
- [x [... cut]
closed 2022-09-22 by mjcarroll