gazebosim/gz-common #351: gz/common/Image.hh does not exist

robberthofmanfm, 2022-05-06:
Trying to compile gazebo garden on Ubuntu 20.04. 

## Environment
* OS Version: Ubuntu 20.04
* Source or binary build?
Source, branch main, latest commit b87a28e

## Description
* Expected behavior: compilation
* Actual behavior: fatal error

## Steps to reproduce

[https://gazebosim.org/docs/garden/install_ubuntu_src](https://gazebosim.org/docs/garden/install_ubuntu_src)

## Output

```
In file included from ~/gazebo_garden_ws/src/ign-common/geospatial/include/ignition/common/geospatial/ImageHeightmap.hh:18,
                 from ~/gazebo_garden_ws/src/ign-common/geospatial/src/ImageHeightmap.cc:18:
~/gazebo_garden_ws/src/ign-common/geospatial/include/gz/common/geospatial/ImageHeightmap.hh:27:10: fatal error: gz/common/Image.hh: No such file or directory
 #include <gz/common/Image.hh>
          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

robberthofmanfm, 2022-05-06:
Both 
`./graphics/include/ignition/common/Image.hh` 
and 
`./graphics/include/gz/common/Image.hh`
exist.

robberthofmanfm, 2022-05-06:
Removed build folder, compiled again, seems to work.
Closing as I don't think it's worth investigating further.

closed 2022-05-06 by robberthofmanfm

Ryanf55, 2024-01-20:
I ran into this today on harmonic. I do not have `libfreeimage-dev` installed. I have gz-sim installed on my OS from binaries, but it's crashing and I wanted a source build in debug to run under GDB. 

A few duplicate files on the system:
```
ryan@B650-970:~/Dev/ros2_ws$ locate Image.hh
/usr/include/gz/common5/gz/common/Image.hh
/usr/include/gz/common5/ignition/common/Image.hh
/usr/include/gz/rendering7/gz/rendering/Image.hh
/usr/include/gz/rendering7/ignition/rendering/Image.hh
/usr/include/gz/rendering8/gz/rendering/Image.hh

ryan@B650-970:~/Dev/ros2_ws$ find src/ -type f -name Image.hh
src/gz-common/graphics/include/ignition/common/Image.hh
src/gz-common/graphics/include/gz/common/Image.hh
src/gz-rendering/include/gz/rendering/Image.hh
```

mjcarroll, 2024-01-20:
> I ran into this today on harmonic. I do not have libfreeimage-dev installed. I have gz-sim installed on my OS from binaries, but it's crashing and I wanted a source build in debug to run under GDB.

So, I think the issue is that `freeimage` is a required dependency of an optional component `graphics`. The `graphics` component is a required dependency of the `geospatial` component.  Perhaps we don't have a way to express this with gz-cmake, but I would expect that the `geospatial` component should also not build in this case.  

This may cause impacts down the line for users who depend on that, so it may make sense to make the configuration warnings a little "louder" in the output logs.

mjcarroll, 2024-01-20:
@Ryanf55 See if this helps: https://github.com/gazebosim/gz-common/pull/573