gazebosim/gz-common #369: Improving Image Data() API to reduce chance of memory leaks

luca-della-vedova, 2022-06-15:
While working on a [PR](https://github.com/gazebosim/gz-common/pull/368) to add support for compressed images to gz-common, the non-ideal behavior of the current API, that is quite prone to memory leaks, came up.

## Desired behavior
It should be safe to use the `Image.Data()` functions without a risk of segfault or memory leaks.

Right now the function takes as an input an `unsigned char **_data` and a `unsigned int& _count` that will be used as output parameters.
There are several vulnerabilities in this API:

`_data` will be freed internally if not equal to nullptr. If the user was to declare but not initialize the variable, i.e.:

```
unsigned char *data;
unsigned int count;
image.Data(&data, count);
```

The program would segfault.

Another issue is that the user needs to know to free manually the data variable after they are done, and again this is quite prone to error.
Finally, the API is slightly confusing since one argument is passed by reference, one by a pointer, and all the parameters are output parameters, it would probably be a bit clearer if it just had a return value.

## Alternatives considered
A possibility could be to create a new Data function that takes no parameters and returns either;
* `std::vector<unsigned char>`: It's great because it can be used out of the box but users have to be a bit careful of copies since the structure containing possibly multiple megabytes of data it can go out of control quite quickly.
* `std::shared_ptr<std::vector<unsigned char>>`: Slightly more work since there is dynamic allocation to do but still memory safe and copies now only change the reference count instead of copying the whole image.

Wildcard for completeness, it seems there is a std::byte [type in C++-17](https://en.cppreference.com/w/cpp/types/byte) which sounds in theory perfect for the purpose, however since downstream users will likely do mathematical operations (i.e. pixelwise operations) and std::byte seems not to allow them, it might be counterproductive.

## Implementation suggestion

My instinct would be to go for the `shared_ptr` approach, returning the vector itself sounds like it would be the cleanest API but when working with downstream packages I noticed that clones and copies happen quite a lot (i.e. before a texture gets rendered by gz-rendering it gets copied at least twice, although I didn't keep track of the destructors to see if it's always cleared).

So my recommendation would be to add new Data() - RGBData() - RGBAData() functions:

```
std::shared_ptr<std::vector<unsigned char>> Data() const;
```

And mark the current

```
void Data(unsigned char **_data, unsigned int &_count) const;
```

as deprecated to be removed in the next release. Then we can refactor (or probably duplicate and deprecate in the same way?) the DataImpl function to be able to return a `shared_ptr`.
We can rely on C++ figuring out from the parameters what function is being called so this behavior should, in theory, allow a full tick/tock cycle without breaking existing behavior and allow users to migrate before Gazebo H.

mjcarroll, 2022-06-21:
> The only case I can think where the output parameter might help is if we do a lot of calls to Data(data) in a row without deallocating the vector and the images are always the same size, imho it's a bit of an obscure case

I thought we made heavier use of that pattern in `gz-sensors` and `gz-sim`, but a quick check shows that I was mistaken.  Given that, I agree that the return value is cleaner.

mjcarroll, 2022-08-09:
Closed by #369

PR #372 (2022-07-26): Add API for getting Image bytes as std::vector
# 🎉 New feature

Closes #369 

## Summary
This PR adds a new API for `common::Image` to get the data as an `std::vector<unsigned char>` and deprecates the API where the user is expected to pass a pointer, a length and free the memory manually.

There is a fair bit of code duplication, I was trying to come up with a way to reuse some code but it would have probably made deprecation harder in the future.
For example I thought of having the std::vector version of the function just copy the raw array returned by the `Data` functions but that would hav [... cut]
closed 2022-08-09 by mjcarroll