gazebosim/gz-common #386: [Garden] AV component dependencies not found in Windows CI

scpeters, 2022-07-03:
## Environment
* OS Version: Windows 10 (CI machines)
* Source or binary build? source build of main branch (garden)

## Description
* Expected behavior: the dependencies should be found by cmake when configuring gz-common5 in CI
  * these machines have `ffmpeg` installed (search for `install external dependency ffmpeg` and `Package ffmpeg:x64-windows is already installed` in console output)
  * ign-common4 still finds these dependencies and runs the `av` tests on these same machines [image] https://build.osrfoundation.org/view/ign-fortress/job/ign_common-ign-4-win/28/testReport/(root)/VideoEncoderTest/
  * gz-common5 was last able to find these dependencies on May 14th https://build.osrfoundation.org/view/ign-garden/job/ign_common-ci-win/42/testReport/(root)/VideoEncoderTest/
 
~~~
-- Found SWSCALE: TRUE  
-- Looking for libswscale - found

-- Found AVDEVICE: TRUE (Required is at least version "56.4.100") 
-- Looking for libavdevice - found

-- Found AVFORMAT: TRUE  
-- Looking for libavformat - found

-- Found AVCODEC: TRUE  
-- Looking for libavcodec - found

-- Found AVUTIL: TRUE  
-- Looking for libavutil - found

~~~

* Actual behavior:
  * the next build after May 14th was on May 18th and it couldn't find the dependencies, though this is not reflected in the build status, since cmake warnings are not reported on windows [image] https://build.osrfoundation.org/view/ign-garden/job/ign_common-ci-win/43/

~~~
[17.704s] -- Could NOT find SWSCALE (missing: SWSCALE_FOUND) 
[17.704s] -- Looking for libswscale - not found
[17.704s] 
[17.813s] -- Could NOT find AVDEVICE (missing: AVDEVICE_FOUND) (Required is at least version "56.4.100")
[17.813s] -- Looking for libavdevice - not found
[17.813s] 
[17.922s] -- Could NOT find AVFORMAT (missing: AVFORMAT_FOUND) 
[17.922s] -- Looking for libavformat - not found
[17.922s] 
[18.032s] -- Could NOT find AVCODEC (missing: AVCODEC_FOUND) 
[18.032s] -- Looking for libavcodec - not found
[18.032s] 
[18.141s] -- Could NOT find AVUTIL (missing: AVUTIL_FOUND) 
[18.141s] -- Looking for libavutil - not found
[18.141s] 
[18.141s] -- -------------------------------------------
[18.141s] 
[18.157s] CMake Warning at C:/Jenkins/workspace/ign_common-ci-win/ws/install/ignition-cmake3/share/cmake/ignition-cmake3/cmake3/IgnConfigureBuild.cmake:69 (message):
[18.157s]    CONFIGURATION WARNINGS:
[18.157s]    -- Skipping component [av]: Missing dependency [libswscale].
[18.157s]       ^~~~~ Set SKIP_av=true in cmake to suppress this warning.
[18.157s]    
[18.157s]    -- Skipping component [av]: Missing dependency [libavdevice].
[18.157s]       ^~~~~ Set SKIP_av=true in cmake to suppress this warning.
[18.157s]    
[18.157s]    -- Skipping component [av]: Missing dependency [libavformat].
[18.157s]       ^~~~~ Set SKIP_av=true in cmake to suppress this warning.
[18.157s]    
[18.157s]    -- Skipping component [av]: Missing dependency [libavcodec].
[18.157s]       ^~~~~ Set SKIP_av=true in cmake to suppress this warning.
[18.157s]    
[18.157s]    -- Skipping component [av]: Missing dependency [libavutil].
[18.157s]       ^~~~~ Set SKIP_av=true in cmake to suppress this warning.
~~~

  * The ign-gazebo7 (aka gz-sim7) builds also started failing on May 18th since the `av` component is required by gz-sim [image] https://build.osrfoundation.org/view/ign-garden/job/ign_gazebo-ci-win/185/

~~~
[16.656s] CMake Warning at C:/vcpkg/scripts/buildsystems/vcpkg.cmake:782 (_find_package):
[16.672s]   By not providing "Findignition-common5-av.cmake" in CMAKE_MODULE_PATH this
[16.672s]   project has asked CMake to find a package configuration file provided by
[16.672s]   "ignition-common5-av", but CMake did not find one.
[16.672s] 
[16.672s]   Could not find a package configuration file provided by
[16.672s]   "ignition-common5-av" (requested version 5.0.0) with any of the following
[16.672s]   names:
[16.672s] 
[16.672s]     ignition-common5-avConfig.cmake
[16.672s]     ignition-common5-av-config.cmake
[16.672s] 
[16.672s]   Add the installation prefix of "ignition-common5-av" to CMAKE_PREFIX_PATH
[16.672s]   or set "ignition-common5-av_DIR" to a directory containing one of the above
[16.672s]   files.  If "ignition-common5-av" provides a separate development package or
[16.672s]   SDK, be sure it has been installed.
...
  -- BUILD ERRORS: These must be resolved before compiling.

CMake Error at C:/Jenkins/workspace/ign_gazebo-ci-win/ws/install/ignition-cmake3/share/cmake/ignition-cmake3/cmake3/IgnConfigureBuild.cmake:75 (message):
  -- 	Missing dependency [ignition-common5] (Components: profiler, events,
  av)
Call Stack (most recent call first):
  C:/Jenkins/workspace/ign_gazebo-ci-win/ws/install/ignition-cmake3/share/cmake/ignition-cmake3/cmake3/IgnConfigureBuild.cmake:39 (gz_configure_build)
  CMakeLists.txt:214 (ign_configure_build)

~~~

## Steps to reproduce

1. Run gz-common5 or gz-sim7 CI builds on Windows
2. check if `av` component is built by gz-common5
3.

## Output

scpeters, 2022-07-03:
my best guess is that this is caused by changes in gz-cmake's `main` branch between May 14-18th. The most likely culprit is https://github.com/gazebosim/gz-cmake/pull/255, which was merged on May 17th, but I haven't been able to identify any issues from looking at those changes again.

scpeters, 2022-07-03:
it is interesting to compare the console output from the first problematic build on May 18th to a more recent build:

* https://build.osrfoundation.org/view/ign-garden/job/ign_common-ci-win/43/

~~~
[17.579s] -- Looking for GDAL - found
[17.579s] 
[17.704s] -- Could NOT find SWSCALE (missing: SWSCALE_FOUND) 
[17.704s] -- Looking for libswscale - not found
[17.704s] 
[17.813s] -- Could NOT find AVDEVICE (missing: AVDEVICE_FOUND) (Required is at least version "56.4.100")
[17.813s] -- Looking for libavdevice - not found
[17.813s] 
[17.922s] -- Could NOT find AVFORMAT (missing: AVFORMAT_FOUND) 
[17.922s] -- Looking for libavformat - not found
[17.922s] 
[18.032s] -- Could NOT find AVCODEC (missing: AVCODEC_FOUND) 
[18.032s] -- Looking for libavcodec - not found
[18.032s] 
[18.141s] -- Could NOT find AVUTIL (missing: AVUTIL_FOUND) 
[18.141s] -- Looking for libavutil - not found
[18.141s] 
[18.141s] -- -------------------------------------------
~~~

* https://build.osrfoundation.org/view/ign-garden/job/ign_common-ci-win/52/

~~~
-- Looking for GDAL - found

CMake Warning (dev) at C:/Jenkins/workspace/ign_common-ci-win/ws/install/gz-cmake3/share/cmake/gz-cmake3/cmake3/IgnUtils.cmake:1932 (message):
  

  The build script has specified some unrecognized arguments for
  gz_import_target(~):

  SWSCALE::SWSCALE

  Either the script has a typo, or it is using an unexpected version of
  gz-cmake.  The version of gz-cmake currently being used is 3.0.0

Call Stack (most recent call first):
  C:/Jenkins/workspace/ign_common-ci-win/ws/install/gz-cmake3/share/cmake/gz-cmake3/cmake3/IgnImportTarget.cmake:69 (_gz_cmake_parse_arguments)
  C:/Jenkins/workspace/ign_common-ci-win/ws/install/gz-cmake3/share/cmake/gz-cmake3/cmake3/IgnManualSearch.cmake:111 (gz_import_target)
  C:/Jenkins/workspace/ign_common-ci-win/ws/install/gz-cmake3/share/cmake/gz-cmake3/cmake3/FindSWSCALE.cmake:23 (gz_manual_search)
  C:/vcpkg/scripts/buildsystems/vcpkg.cmake:782 (_find_package)
  C:/Jenkins/workspace/ign_common-ci-win/ws/install/gz-cmake3/share/cmake/gz-cmake3/cmake3/IgnUtils.cmake:212 (find_package)
  CMakeLists.txt:109 (gz_find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Could NOT find SWSCALE (missing: SWSCALE_FOUND) 
-- Looking for libswscale - not found

CMake Warning (dev) at C:/Jenkins/workspace/ign_common-ci-win/ws/install/gz-cmake3/share/cmake/gz-cmake3/cmake3/IgnUtils.cmake:1932 (message):
  

  The build script has specified some unrecognized arguments for
  gz_import_target(~):

  AVDEVICE::AVDEVICE

  Either the script has a typo, or it is using an unexpected version of
  gz-cmake.  The version of gz-cmake currently being used is 3.0.0

Call Stack (most recent call first):
  C:/Jenkins/workspace/ign_common-ci-win/ws/install/gz-cmake3/share/cmake/gz-cmake3/cmake3/IgnImportTarget.cmake:69 (_gz_cmake_parse_arguments)
  C:/Jenkins/workspace/ign_common-ci-win/ws/install/gz-cmake3/share/cmake/gz-cmake3/cmake3/IgnManualSearch.cmake:111 (gz_import_target)
  C:/Jenkins/workspace/ign_common-ci-win/ws/install/gz-cmake3/share/cmake/gz-cmake3/cmake3/FindAVDEVICE.cmake:27 (gz_manual_search)
  C:/vcpkg/scripts/buildsystems/vcpkg.cmake:782 (_find_package)
  C:/Jenkins/workspace/ign_common-ci-win/ws/install/gz-cmake3/share/cmake/gz-cmake3/cmake3/IgnUtils.cmake:212 (find_package)
[... 52 lines cut ...]
CMake Warning (dev) at C:/Jenkins/workspace/ign_common-ci-win/ws/install/gz-cmake3/share/cmake/gz-cmake3/cmake3/IgnUtils.cmake:1932 (message):
  

  The build script has specified some unrecognized arguments for
  gz_import_target(~):

  AVUTIL::AVUTIL

  Either the script has a typo, or it is using an unexpected version of
  gz-cmake.  The version of gz-cmake currently being used is 3.0.0

Call Stack (most recent call first):
  C:/Jenkins/workspace/ign_common-ci-win/ws/install/gz-cmake3/share/cmake/gz-cmake3/cmake3/IgnImportTarget.cmake:69 (_gz_cmake_parse_arguments)
  C:/Jenkins/workspace/ign_common-ci-win/ws/install/gz-cmake3/share/cmake/gz-cmake3/cmake3/IgnManualSearch.cmake:111 (gz_import_target)
  C:/Jenkins/workspace/ign_common-ci-win/ws/install/gz-cmake3/share/cmake/gz-cmake3/cmake3/FindAVUTIL.cmake:23 (gz_manual_search)
  C:/vcpkg/scripts/buildsystems/vcpkg.cmake:782 (_find_package)
  C:/Jenkins/workspace/ign_common-ci-win/ws/install/gz-cmake3/share/cmake/gz-cmake3/cmake3/IgnUtils.cmake:212 (find_package)
  CMakeLists.txt:125 (gz_find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Could NOT find AVUTIL (missing: AVUTIL_FOUND) 
-- Looking for libavutil - not found

-- -------------------------------------------
~~~

scpeters, 2022-07-03:
Potential fix in https://github.com/gazebosim/gz-cmake/commit/7b3cd52dbea18bfaae4f9d709adc75092b2ec8de, testing in https://github.com/gazebosim/gz-common/pull/387 using `ci_matching_branch` with https://github.com/gazebo-tooling/gazebodistro/commit/3d2068306c89bfb9144f7689a4119ddcc124df42

mjcarroll, 2022-08-31:
Closed by https://github.com/gazebosim/gz-cmake/pull/271

closed 2022-08-31 by mjcarroll