gazebosim/gz-common #388: Make the dependency on FreeImage optional

mahiuchun, 2022-07-08:
## Desired behavior

gz-common could be built without using FreeImage.

## Alternatives considered

N/A

## Implementation suggestion

Use https://github.com/nothings/stb/blob/master/stb_image.h for image loading.

## Additional context

- By not supporting less commonly used image formats, the attack surface of image loading is reduced.
- FreeImage has an active SVN but they have not made a release since 2018.

wentasah, 2024-07-15:
This might be a good idea, as freeimage [is known to contain](https://github.com/NixOS/nixpkgs/issues/298114#issuecomment-2227309103) many [security vulnerabilities](https://github.com/NixOS/nixpkgs/blob/7e7c39ea35c5cdd002cd4588b03a3fb9ece6fad9/pkgs/development/libraries/freeimage/default.nix#L75).

mjcarroll, 2024-07-15:
Yes, there has been a little bit of work attempting to replace the dependency with something like `stb` (https://github.com/gazebosim/gz-common/pull/590)  but I haven't had bandwidth to continue on it.

oysstu, 2025-08-13:
Is QImageReader a possible alternative? A downside is that it pulls not only qt core but also qt gui.

mjcarroll, 2025-08-13:
> Is QImageReader a possible alternative? A downside is that it pulls not only qt core but also qt gui.

We don't have Qt as a dependency this low in the stack.  I would think something like `stb` or an equivalent lightweight loader.  I also saw that someone created https://github.com/agruzdev/FreeImageRe

I think it's binary compatible and fixes the CVEs.

oysstu, 2025-08-13:
> We don't have Qt as a dependency this low in the stack. I would think something like `stb` or an equivalent lightweight loader. I also saw that someone created [...]

Figured that would be the case. Looks like FreeImageRe has been under fairly active development by two devs for the past two years. Stb looks decent also, if the supported formats are sufficient. ImageMagick is also an alternative with good format support. Perhaps not lightweight though.

scpeters, 2026-03-13:
rather than make it optional, we are replacing the dependency on freeimage with stb (see https://github.com/gazebosim/gz-common/issues/771 and many thanks to @wentasah!)