gazebosim/gz-common #390: Description of the GitHub repo still refers to Ignition

traversaro, 2022-07-11:
## Environment
* OS Version: Not Relevant
* Source or binary build? Not Relevant

## Description

* Expected behavior: The description of the GitHub repo does not contain the old "Ignition" string.
* Actual behavior: The description of the repo is "An audio-visual library supports processing audio and video files, a graphics library can load a variety 3D mesh file formats into a generic in-memory representation, and the core library of Ignition Common contains functionality that spans Base64 encoding/decoding to thread pools.", i.e. it still refers to igniton.

## Steps to reproduce

Check the description.

## Output

Not relevnat.

mjcarroll, 2022-07-11:
I took a look and I don't have sufficient permissions.  Probably needs someone with owner-level.  @chapulina can you take care of this one?

chapulina, 2022-07-11:
Updated all `gz-` repositories

closed 2022-07-11 by chapulina