gazebosim/gz-common #430: tinyxml2 in source seems to conflict with system tinyxml2

julianoes, 2022-08-25:
## Environment
* OS Version: Debian 11 Bullseye
* Source build using colcon as documented in https://gazebosim.org/docs/fortress/install_ubuntu_src

## Description

I'm trying to build Ignition Gazebo 6 on Debian. The colcon build fails with a linking error for gz-common. It looks like something goes wrong with the linking of tinyxml2 symbols. Presumably the tinyxml2 library installed on my system conflicts with tinyxml2 in tree here.

I tried to build everything without having tinyxml2 installed system-wide but that does not work either as it is required by sdformat12.

## Steps to reproduce

```
vcs import < collection-fortress.yaml
colcon graph
colcon build --cmake-args -DBUILD_TESTING=OFF --merge-install
```

Should I provide a debian based Dockerfile to reproduce it?

## Output

Colcon build without `libtinyxml2-dev` installed system-wide:
```
Starting >>> sdformat12                                                              
--- stderr: sdformat12                                                 
CMake Error at /home/julianoes/src/other/ign-workspace/install/share/cmake/ignition-cmake2/cmake2/IgnConfigureBuild.cmake:59 (message):
  -- BUILD ERRORS: These must be resolved before compiling.
Call Stack (most recent call first):
  CMakeLists.txt:115 (ign_configure_build)

CMake Error at /home/julianoes/src/other/ign-workspace/install/share/cmake/ignition-cmake2/cmake2/IgnConfigureBuild.cmake:61 (message):
  -- 	Missing dependency [TINYXML2]
Call Stack (most recent call first):
  CMakeLists.txt:115 (ign_configure_build)
```

Build with `tinyxml2` installed:

```
/usr/bin/ld: CMakeFiles/sdformat12.dir/urdf/urdf_parser/model.cpp.o: in function `tinyxml2::XMLDocument::ClearError()':
/usr/include/tinyxml2.h:1871: undefined reference to `tinyxml2::XMLDocument::SetError(tinyxml2::XMLError, int, char const*, ...)'
/usr/bin/ld: ../lib/libsdformat12.so.12.5.0: hidden symbol `_ZN8tinyxml211XMLDocument8SetErrorENS_8XMLErrorEiPKcz' isn't defined
/usr/bin/ld: final link failed: bad value
collect2: error: ld returned 1 exit status
gmake[2]: *** [src/CMakeFiles/sdformat12.dir/build.make:1247: lib/libsdformat12.so.12.5.0] Error 1
```

mjcarroll, 2022-08-31:
>  The colcon build fails with a linking error for gz-common. 

Looking at the errors you included here, it looks like entirely sdformat issues. Do you have the additional `gz-common` errors as well?

julianoes, 2022-08-31:
@mjcarroll you're probably right that it's a sdformat issue, however, it's still connected to the fact that tinyxml2 shows up in source here, and might be playing a role.

I'll open an issue for sdformat then as well.

mjcarroll, 2022-09-16:
Closed in favor of gazebosim/sdformat#1124

closed 2022-09-16 by mjcarroll