gazebosim/gz-common #450: Failure in ignition-common3 deb packages served by packages.osrfoundation.org

traversaro, 2022-10-09:
Since the last ~2 days, installing `libgazebo11-dev` on 20.04 from packages.osrfoundation.org fails with the following error messages:

~~~
2022-10-09T02:58:03.1690968Z Unpacking libignition-common3-av-dev:amd64 (3.15.0-1~focal) ...
2022-10-09T02:58:03.1728592Z dpkg: error processing archive /tmp/apt-dpkg-install-4SWfIT/172-libignition-common3-av-dev_3.15.0-1~focal_amd64.deb (--unpack):
2022-10-09T02:58:03.1729399Z  trying to overwrite '/usr/include/ignition/common3/gz/common/av.hh', which is also in package libignition-common3-core-dev:amd64 3.15.0-1~focal
[...]
2022-10-09T02:58:03.3748468Z Unpacking libignition-common3-events-dev:amd64 (3.15.0-1~focal) ...
2022-10-09T02:58:03.3782141Z dpkg: error processing archive /tmp/apt-dpkg-install-4SWfIT/176-libignition-common3-events-dev_3.15.0-1~focal_amd64.deb (--unpack):
2022-10-09T02:58:03.3783299Z  trying to overwrite '/usr/include/ignition/common3/gz/common/events.hh', which is also in package libignition-common3-core-dev:amd64 3.15.0-1~focal
[...]
2022-10-09T02:58:03.6595796Z Unpacking libignition-common3-graphics-dev:amd64 (3.15.0-1~focal) ...
2022-10-09T02:58:03.6630460Z dpkg: error processing archive /tmp/apt-dpkg-install-4SWfIT/181-libignition-common3-graphics-dev_3.15.0-1~focal_amd64.deb (--unpack):
2022-10-09T02:58:03.6631311Z  trying to overwrite '/usr/include/ignition/common3/gz/common/graphics.hh', which is also in package libignition-common3-core-dev:amd64 3.15.0-1~focal
[...]
2022-10-09T02:58:14.8550652Z Unpacking libgazebo11-dev:amd64 (11.12.0-1~focal) ...
2022-10-09T02:58:15.0408227Z Errors were encountered while processing:
2022-10-09T02:58:15.0409062Z  /tmp/apt-dpkg-install-4SWfIT/172-libignition-common3-av-dev_3.15.0-1~focal_amd64.deb
2022-10-09T02:58:15.0409689Z  /tmp/apt-dpkg-install-4SWfIT/176-libignition-common3-events-dev_3.15.0-1~focal_amd64.deb
2022-10-09T02:58:15.0410335Z  /tmp/apt-dpkg-install-4SWfIT/181-libignition-common3-graphics-dev_3.15.0-1~focal_amd64.deb
2022-10-09T02:58:15.0899871Z E: Sub-process /usr/bin/dpkg returned an error code (1)
2022-10-09T02:58:15.0976735Z ##[error]Process completed with exit code 100.
~~~

Full log at https://github.com/robotology/robotology-superbuild/actions/runs/3212647928 .

## Environment
* OS Version: Ubuntu 20.04 
* Source or binary build? Binary 3.15.0
  
  

## Description
* Expected behavior: 
* Actual behavior: 

## Steps to reproduce

1.
2.
3.

## Output

traversaro, 2022-10-09:
I realized only later that this is probably a duplicate of https://github.com/osrf/gazebo/issues/3266 .

MatthijsBurgh, 2022-10-10:
@traversaro Yes, similar to the issue you mentioned. Though I think this repo is the better location for this issue. As it is not related to gazebo, but the ignition libs.

Aladdin-Ark, 2022-10-10:
@traversaro I can confirm the problem is fixed on my side with the release of `3.15.0-3`

scpeters, 2022-10-10:
this should be fixed since we rebuilt after https://github.com/gazebo-release/gz-common3-release/pull/8 was merged

closed 2022-10-10 by scpeters