gazebosim/gz-common #505: EnumIterator inherits from std::iterator base class which is deprecated in c++17

scpeters, 2023-04-17:
## Environment
* OS Version: macOS 12.6.5 (Monterey) with macOS 13.1 SDK
* Source or binary build? source build of gz-common5 as of 6f5ee941536e5c781bd826f6ce35360347d581ba (other versions too)

excerpt of `brew config`:

~~~
Clang: 14.0.0 build 1400
macOS: 12.6.5-x86_64
CLT: 14.2.0.0.1.1668646533
Xcode: N/A
~~~

## Description
* Expected behavior: package should compile without warnings
* Actual behavior: deprecation warning from `EnumIface.hh` when compiling `EnumIface_TEST.cc` since [`EnumIterator` inherits from `std::iterator`](https://github.com/gazebosim/gz-common/blob/gz-common5_5.3.1/include/gz/common/EnumIface.hh#L143-L145) and the `std::iterator` base class is [deprecated as of c++17](https://en.cppreference.com/w/cpp/iterator)

[image]

~~~
src/EnumIface_TEST.cc:20:
/include/gz/common/EnumIface.hh:145:12: warning: 'iterator<std::bidirectional_iterator_tag, MyType>' is deprecated [-Wdeprecated-declarations]
    : std::iterator<std::bidirectional_iterator_tag, Enum>
           ^
src/EnumIface_TEST.cc:61:32: note: in instantiation of template class 'gz::common::EnumIterator<MyType>' requested here
  common::EnumIterator<MyType> end = MY_TYPE_END;
                               ^
/Library/Developer/CommandLineTools/SDKs/MacOSX13.1.sdk/usr/include/c++/v1/__iterator/iterator.h:27:29: note: 'iterator<std::bidirectional_iterator_tag, MyType>' has been explicitly marked deprecated here
struct _LIBCPP_TEMPLATE_VIS _LIBCPP_DEPRECATED_IN_CXX17 iterator
                            ^
/Library/Developer/CommandLineTools/SDKs/MacOSX13.1.sdk/usr/include/c++/v1/__config:1066:39: note: expanded from macro '_LIBCPP_DEPRECATED_IN_CXX17'
#  define _LIBCPP_DEPRECATED_IN_CXX17 _LIBCPP_DEPRECATED
                                      ^
/Library/Developer/CommandLineTools/SDKs/MacOSX13.1.sdk/usr/include/c++/v1/__config:1043:48: note: expanded from macro '_LIBCPP_DEPRECATED'
#    define _LIBCPP_DEPRECATED __attribute__ ((deprecated))
                                               ^
~~~

## Steps to reproduce

1. Build from source on macOS Monterey

or

1. Run https://build.osrfoundation.org/view/ign-garden/job/ignition_common-ci-gz-common5-homebrew-amd64 on one of the macOS-monterey nodes

## Output

[image] https://build.osrfoundation.org/view/ign-garden/job/ignition_common-ci-gz-common5-homebrew-amd64/37/clang/

scpeters, 2023-04-21:
@mjcarroll is there a way to fix this on released versions without breaking API? If not, I think we should suppress the warning and fix it on `main`

azeey, 2023-05-01:
I've added the `harmonic` label to make sure we include this in Harmonic.

mjcarroll, 2023-05-02:
> I think we should suppress the warning and fix it on main

We concluded this is the right answer after talking about it at the meeting yesterday.

scpeters, 2023-07-29:
@mjcarroll did you have any ideas for how to fix this on `main`?

scpeters, 2023-09-28:
fixed in https://github.com/gazebosim/gz-common/pull/529

PR #529 (2023-09-28): Remove deprecated std::iterator inheritance
# 🦟 Bug fix

Fixes #505

## Summary

Removes the use of the deprecated std::iterator template and replaces it with using the correct set of `using` statements.

I also added a `gzEnum` variadic template at clang's suggestion.  This should probably be preferred over the macro where relevant.

## Checklist
- [ ] Signed all commits for DCO
- [ ] Added tests
- [ ] Updated documentation (as needed)
- [ ] Updated migration guide (as needed)
- [ ] Consider updating Python bindings (if the library has them)
- [ ] `codecheck` passed (See [contributing](https: [... cut]
closed 2023-09-28 by scpeters