gazebosim/gz-common #520: How to create a Mesh using Vertices and Triangle indices ?

Levi-Armstrong, 2023-06-15:
I am creating a visualization tool using GzRendering which runs convex hull decomposition and I want to create a mesh using Vertices and Triangle Indices provide by these tools and visualize. How can this be accomplished without writing to a file and loading?

iche033, 2023-06-16:
I don't think we have a tutorial for this but you can fine example code here for creating a box mesh: https://github.com/gazebosim/gz-common/blob/gz-common5/graphics/src/MeshManager.cc#L458

After that you can pass the mesh object to gz-rendering like so: https://github.com/gazebosim/gz-rendering/blob/gz-rendering7/examples/mesh_viewer/Main.cc#L59-L68 

In the above example, you'll just need to set the mesh directly to the one you created instead of asking the MeshManager to load the mesh for you, i.e.

change 

```cpp
descriptor.mesh = meshManager->Load(descriptor.meshName);
```
to

```cpp
descriptor.mesh = your_mesh_obj
```

azeey, 2023-06-26:
Closing since the feature is already available.

closed 2023-06-26 by azeey

Levi-Armstrong, 2023-06-30:
@iche033 is it required to provide normal's?

Levi-Armstrong, 2023-07-01:
It appears normal's are required for things to visual correctly, but after looking at the provided links and the API for the SubMesh I was able to get things working. @iche033 Thanks for the support.