gazebosim/gz-common #522: gz-common5 compile error: ColladaLoader.hh: No such file or directory

Ryanf55, 2023-07-07:
## Environment
* OS Version: Ubuntu 23.04
* Source or binary build?
Source, commit 6a0f9d7cd67a3916a0e6661cd99494f2d0e4faf9

## Description
* Expected behavior: Can build successfully with colcon
* Actual behavior: Include error

## Steps to reproduce

Follow the instructions for Ubuntu source build on Ubuntu 23.04, except install dependencies manually with apt. 
1. sudo apt install libprotobuf-dev libprotoc-dev protobuf-compiler
2. colcon build --packages-up-to gz-common
3. 
4.

## Output

Here's the CMake configure, which is basically the same as colcon. I've already built the dependencies.
```
ryan@ryan-B650-970:~/Development/gz_garden/src/gz-common$ cmake -S . -B build
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/lib/ccache/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/lib/ccache/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- gz-common5 version 5.4.0
-- Operating system is Linux
-- Found CPack generators: DEB
-- 

-- ====== Finding Dependencies ======
-- Looking for gz-math7 -- found version 7.2.0
-- Searching for dependencies of gz-math7
-- Looking for gz-utils2 -- found version 2.0.0
-- Searching for dependencies of gz-utils2
-- Looking for gz-math7 - found

-- Looking for gz-utils2 -- found version 2.0.0
-- Looking for gz-utils2 - found

-- Checking for module 'tinyxml2'
--   Found tinyxml2, version 9.0.0
-- Looking for tinyxml2 - found

-- Looking for dlfcn.h - found
-- Looking for libdl - found
-- Found DL: TRUE  
-- Looking for libdl - found

-- Checking for module 'uuid'
--   Found uuid, version 2.38.1
-- Found UUID: TRUE  
-- Looking for uuid - found

-- Looking for FreeImage.h - not found
-- Missing: Unable to find FreeImage.h
-- Missing: Unable to find libfreeimage
-- Could NOT find FreeImage (missing: FreeImage_FOUND) (Required is at least version "3.9")
-- Looking for FreeImage - not found

-- Checking for module 'gts'
--   Package 'gts', required by 'virtual:world', not found
-- Could NOT find GTS (missing: GTS_FOUND) 
-- Looking for gts - not found

-- Could NOT find GDAL: Found unsuitable version "GDAL_VERSION-NOTFOUND", but required is at least "3.0" (found GDAL_LIBRARY-NOTFOUND)
-- Looking for GDAL - not found

-- Checking for module 'libswscale'
--   Found libswscale, version 6.7.100
-- Found SWSCALE: TRUE  
-- Looking for libswscale - found

-- Checking for module 'libavdevice >= 56.4.100'
--   Package 'libavdevice', required by 'virtual:world', not found
-- Could NOT find AVDEVICE (missing: AVDEVICE_FOUND) (Required is at least version "56.4.100")
-- Looking for libavdevice - not found

-- Checking for module 'libavformat'
--   Found libavformat, version 59.27.100
-- Found AVFORMAT: TRUE  
-- Looking for libavformat - found

-- Checking for module 'libavcodec'
--   Found libavcodec, version 59.37.100
-- Found AVCODEC: TRUE  
-- Looking for libavcodec - found

-- Checking for module 'libavutil'
--   Found libavutil, version 57.28.100
-- Found AVUTIL: TRUE  
-- Looking for libavutil - found

-- Found GzAssimp: TRUE (found version "5.2.4") 
-- Looking for assimp - found

-- -------------------------------------------

CMake Warning at /home/ryan/Development/gz_garden/install/gz-cmake3/share/cmake/gz-cmake3/cmake3/GzConfigureBuild.cmake:68 (message):
   CONFIGURATION WARNINGS:
   -- Skipping component [graphics]: Missing dependency [FreeImage].
      ^~~~~ Set SKIP_graphics=true in cmake to suppress this warning.
   
   -- Skipping component [graphics]: Missing dependency [gts] - GNU Triangulation Surface library.
      ^~~~~ Set SKIP_graphics=true in cmake to suppress this warning.
   
   -- Skipping component [geospatial]: Missing dependency [GDAL].
      ^~~~~ Set SKIP_geospatial=true in cmake to suppress this warning.
   
   -- Skipping component [av]: Missing dependency [libavdevice].
      ^~~~~ Set SKIP_av=true in cmake to suppress this warning.
   
Call Stack (most recent call first):
  CMakeLists.txt:139 (gz_configure_build)

-- 
-- Searching for host SSE information
-- SSE2 found
-- SSE3 found
-- SSE4.1 found
-- SSE4.2 found
-- Configuring library: gz-common5
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY - Success
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY - Success
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR - Success
-- Adding 20 UNIT tests
-- Adding 4 INTEGRATION tests
-- Adding 2 PERFORMANCE tests
-- No tests have been specified for REGRESSION
-- Performing Test Rflag
-- Performing Test Rflag - Success
-- Skipping the component [av] because the following packages are missing: libavdevice
-- Configuring library: gz-common5-events
-- Adding 3 UNIT tests
-- Skipping the component [geospatial] because the following packages are missing: GDAL
-- Skipping the component [graphics] because the following packages are missing: FreeImage, gts
-- Configuring library: gz-common5-io
-- Adding 2 UNIT tests
-- Configuring library: gz-common5-profiler
-- Adding 3 UNIT tests
-- Configuring library: gz-common5-testing
-- Adding 4 UNIT tests
-- Adding codecheck target
-- Build configuration successful
-- Build type: RelWithDebInfo
-- Install prefix: /usr/local
-- Found Doxygen: /usr/bin/doxygen (found version "1.9.4") found components: doxygen dot 
-- Configuring done (1.8s)
-- Generating done (0.1s)
-- Build files have been written to: /home/ryan/Development/gz_garden/src/gz-common/build
```

Here's the CMake build 
```
ryan@ryan-B650-970:~/Development/gz_garden/src/gz-common$ cmake --build build
[  1%] Generating API documentation with Doxygen
warning: Tag 'OUTPUT_TEXT_DIRECTION' at line 102 of file '/home/ryan/Development/gz_garden/src/gz-common/build/api_tagfile.dox' has become obsolete.
...
[ 57%] Building CXX object test/integration/CMakeFiles/INTEGRATION_deprecated_TEST.dir/deprecated_TEST.cc.o
[ 58%] Linking CXX executable ../../bin/INTEGRATION_deprecated_TEST
[ 58%] Built target INTEGRATION_deprecated_TEST
[ 59%] Building CXX object test/integration/CMakeFiles/INTEGRATION_mesh.dir/mesh.cc.o
/home/ryan/Development/gz_garden/src/gz-common/test/integration/mesh.cc:21:10: fatal error: gz/common/ColladaLoader.hh: No such file or directory
   21 | #include "gz/common/ColladaLoader.hh"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [test/integration/CMakeFiles/INTEGRATION_mesh.dir/build.make:76: test/integration/CMakeFiles/INTEGRATION_mesh.dir/mesh.cc.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:2226: test/integration/CMakeFiles/INTEGRATION_mesh.dir/all] Error 2
gmake: *** [Makefile:166: all] Error 2
```

mjcarroll, 2023-07-10:
This is a bug.  You are missing `gts` so the `graphics` component isn't being built.  Because that component is missing, the corresponding integration tests should not have been built.

mjcarroll, 2023-07-12:
Can you give #523 a look and see if it resolves this for you?

mjcarroll, 2023-07-12:
Just out of curiosity, are you intentionally avoiding these dependencies or had you just not installed them yet at this point?

Ryanf55, 2023-07-12:
> Just out of curiosity, are you intentionally avoiding these dependencies or had you just not installed them yet at this point?

Just hadn't installed it, I was going through all the `find_package` errors to install each dependency, but this one manifested the wrong way.

mjcarroll, 2023-07-12:
Sounds good, if it is helpful, we have included files in each repo to support CI workflows, maybe helpful in also finding dependencies (eg https://github.com/gazebosim/gz-common/blob/gz-common5/.github/ci/packages.apt)

We aren't using package.xml for historical reasons that haven't been revisited recently.

Ryanf55, 2023-07-12:
Thanks, I'll take a look at those.

PR #523 (2023-07-12): Only build mesh integration tests if graphics libraries are built
# 🦟 Bug fix

Fixes #522 

## Summary

This removes `mesh.cc` if the graphics component is not built either by being disabled by the user or from missing dependencies.

## Checklist
- [ ] Signed all commits for DCO
- [ ] Added tests
- [ ] Updated documentation (as needed)
- [ ] Updated migration guide (as needed)
- [ ] Consider updating Python bindings (if the library has them)
- [ ] `codecheck` passed (See [contributing](https://gazebosim.org/docs/all/contributing#contributing-code))
- [ ] All tests passed (See [test coverage](h [... cut]
closed 2023-07-12 by mjcarroll