gazebosim/gz-common #544: VideoEncoder does not encode to v4l2 loopback device

vryabokon1705, 2023-10-16:
## Environment
* OS version: Ubuntu 22.04
* Source or binary build: binary
  Gz-common: 5.4.2-1
  Gz-sim: 7.6.0-1

## Description
* Expected behavior: v4l2 loopback device simulates video from world`s camera
* Actual behavior: Simulation fails with error log message: "[Err] [VideoEncoder.cc:435] Unable to allocate format context. Video encoding not started" 

## Steps to reproduce

1. Build fork of regular CameraVideoRecorder plugin with possibility to use v4l2 format : https://github.com/vryabokon1705/camera-videro-recorder2.git
2. Copy libCameraVideoRecorder2.so to $HOME/.gz/sim/plugins/
3. Make copy of camera_video_record_dbl_pendulum.sdf as camera_video_record_dbl_pendulum_copy.sdf
4. Edit camera_video_record_dbl_pendulum_copy.sdf: 
`<sensor name="camera" type="camera">`
         `<plugin filename="CameraVideoRecorder2" name="streamer::CameraVideoRecorder2">`
            `<service>camera/record_video</service>`
          `</plugin>`
`...`
`</sensor>`
5. Check existing v4l2 devices 
`ls /dev/video*`
6. Create v4l2 loopback device: 
`sudo modprobe v4l2loopback`
find new created v4l2 loopback device path (let it be /dev/video1)
`ls /dev/video*`
7. Start gazebo simulator:
`gz sim camera_video_record_dbl_pendulum.sdf`
8. Start gazebo service: 
`gz service -s /camera/record_video1 --reqtype gz.msgs.VideoRecord --reptype gz.msgs.Boolean --timeout 300 --req 'start: true, format:"v4l2", save_filename:"/dev/video1"'`

## Output
[image]
[image]

vryabokon1705, 2023-10-17:
Looks like string 349
`this->dataPtr->format = _format.compare("v4l") == 0 ? "v4l2" : _format; `
must be replaced with 
`this->dataPtr->format = _format.compare("v4l") == 0 || _format.compare("v4l2") == 0 ? "video4linux2,v4l2" : _format;`

mjcarroll, 2023-11-06:
@vryabokon1705 any chance you could open a PR with that?