gazebosim/gz-common #579: Build error ignition-common3_3.17 with ffmpeg-5.0

bchoineubility, 2024-03-07:
Hi, 

I had faced build error for ignition-common3_3.17 in Yocto.

log is as following.

```
 [13/64] /home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a-oe4t-linux/ignition-common3/3.17.0-r0/recipe-sysroot-native/usr/bin/aarch64-oe4t-linux/aarch64-oe4t-linux-g++ --sysroot=/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a-oe4t-linux/ignition-common3/3.17.0-r0/recipe-sysroot -Dignition_common3_EXPORTS -I/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a-oe4t-linux/ignition-common3/3.17.0-r0/git/include -I/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a-oe4t-linux/ignition-common3/3.17.0-r0/build/include -I/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a-oe4t-linux/ignition-common3/3.17.0-r0/build/core/include -I/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a-oe4t-linux/ignition-common3/3.17.0-r0/recipe-sysroot/usr/include/uuid -march=armv8-a+crc -fstack-protector-strong  -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a-oe4t-linux/ignition-common3/3.17.0-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a-oe4t-linux/ignition-common3/3.17.0-r0=/usr/src/debug/ignition-common3/3.17.0-r0                      -fdebug-prefix-map=/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a-oe4t-linux/ignition-common3/3.17.0-r0=/usr/src/debug/ignition-common3/3.17.0-r0                      -fdebug-prefix-map=/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a-oe4t-linux/ignition-common3/3.17.0-r0/recipe-sysroot=                      -fdebug-prefix-map=/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a-oe4t-linux/ignition-common3/3.17.0-r0/recipe-sysroot-native=  -fvisibility-inlines-hidden -O2 -g -DNDEBUG  -Wall -Wextra -Wno-long-long -Wno-unused-value -Wfloat-equal -Wshadow -Winit-self -Wswitch-default -Wmissing-include-dirs -pedantic -UNDEBUG -fPIC -fvisibility=default -fPIC -I/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a-oe4t-linux/ignition-common3/3.17.0-r0/recipe-sysroot/usr/include/uuid -std=c++17 -MD -MT src/CMakeFiles/ignition-common3.dir/PluginLoader.cc.o -MF src/CMakeFiles/ignition-common3.dir/PluginLoader.cc.o.d -o src/CMakeFiles/ignition-common3.dir/PluginLoader.cc.o -c /home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a-oe4t-linux/ignition-common3/3.17.0-r0/git/src/PluginLoader.cc
| [14/64] /home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a-oe4t-linux/ignition-common3/3.17.0-r0/recipe-sysroot-native/usr/bin/aarch64-oe4t-linux/aarch64-oe4t-linux-g++ --sysroot=/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a-oe4t-linux/ignition-common3/3.17.0-r0/recipe-sysroot -DIGN_COMMON_BUILD_HW_VIDEO -Dignition_common3_av_EXPORTS -I/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a-oe4t-linux/ignition-common3/3.17.0-r0/git/include -I/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a-oe4t-linux/ignition-common3/3.17.0-r0/build/include -I/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a-oe4t-linux/ignition-common3/3.17.0-r0/git/av/include -I/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a-oe4t-linux/ignition-common3/3.17.0-r0/build/av/include -I/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a-oe4t-linux/ignition-common3/3.17.0-r0/build/core/include -I/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a-oe4t-linux/ignition-common3/3.17.0-r0/recipe-sysroot/usr/include/uuid -march=armv8-a+crc -fstack-protector-strong  -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a-oe4t-linux/ignition-common3/3.17.0-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a-oe4t-linux/ignition-common3/3.17.0-r0=/usr/src/debug/ignition-common3/3.17.0-r0                      -fdebug-prefix-map=/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a-oe4t-linux/ignition-common3/3.17.0-r0=/usr/src/debug/ignition-common3/3.17.0-r0                      -fdebug-prefix-map=/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a-oe4t-linux/ignition-common3/3.17.0-r0/recipe-sysroot=                      -fdebug-prefix-map=/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a-oe4t-linux/ignition-common3/3.17.0-r0/recipe-sysroot-native=  -fvisibility-inlines-hidden -O2 -g -DNDEBUG  -Wall -Wextra -Wno-long-long -Wno-unused-value -Wfloat-equal -Wshadow -Winit-self -Wswitch-default -Wmissing-include-dirs -pedantic -UNDEBUG -fPIC -fvisibility=default -fPIC -I/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a-oe4t-linux/ignition-common3/3.17.0-r0/recipe-sysroot/usr/include/uuid -std=c++17 -MD -MT av/src/CMakeFiles/ignition-common3-av.dir/AudioDecoder.cc.o -MF av/src/CMakeFiles/ignition-common3-av.dir/AudioDecoder.cc.o.d -o av/src/CMakeFiles/ignition-common3-av.dir/AudioDecoder.cc.o -c /home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a-oe4t-linux/ignition-common3/3.17.0-r0/git/av/src/AudioDecoder.cc
| FAILED: av/src/CMakeFiles/ignition-common3-av.dir/AudioDecoder.cc.o
| /home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a-oe4t-linux/ignition-common3/3.17.0-r0/recipe-sysroot-native/usr/bin/aarch64-oe4t-linux/aarch64-oe4t-linux-g++ --sysroot=/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a-oe4t-linux/ignition-common3/3.17.0-r0/recipe-sysroot -DIGN_COMMON_BUILD_HW_VIDEO -Dignition_common3_av_EXPORTS -I/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a-oe4t-linux/ignition-common3/3.17.0-r0/git/include -I/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a-oe4t-linux/ignition-common3/3.17.0-r0/build/include -I/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a-oe4t-linux/ignition-common3/3.17.0-r0/git/av/include -I/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a-oe4t-linux/ignition-common3/3.17.0-r0/build/av/include -I/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a-oe4t-linux/ignition-common3/3.17.0-r0/build/core/include -I/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a-oe4t-linux/ignition-common3/3.17.0-r0/recipe-sysroot/usr/include/uuid -march=armv8-a+crc -fstack-protector-strong  -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a-oe4t-linux/ignition-common3/3.17.0-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a-oe4t-linux/ignition-common3/3.17.0-r0=/usr/src/debug/ignition-common3/3.17.0-r0                      -fdebug-prefix-map=/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a-oe4t-linux/ignition-common3/3.17.0-r0=/usr/src/debug/ignition-common3/3.17.0-r0                      -fdebug-prefix-map=/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a-oe4t-linux/ignition-common3/3.17.0-r0/recipe-sysroot=                      -fdebug-prefix-map=/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a-oe4t-linux/ignition-common3/3.17.0-r0/recipe-sysroot-native=  -fvisibility-inlines-hidden -O2 -g -DNDEBUG  -Wall -Wextra -Wno-long-long -Wno-unused-value -Wfloat-equal -Wshadow -Winit-self -Wswitch-default -Wmissing-include-dirs -pedantic -UNDEBUG -fPIC -fvisibility=default -fPIC -I/home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a-oe4t-linux/ignition-common3/3.17.0-r0/recipe-sysroot/usr/include/uuid -std=c++17 -MD -MT av/src/CMakeFiles/ignition-common3-av.dir/AudioDecoder.cc.o -MF av/src/CMakeFiles/ignition-common3-av.dir/AudioDecoder.cc.o.d -o av/src/CMakeFiles/ignition-common3-av.dir/AudioDecoder.cc.o -c /home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a-oe4t-linux/ignition-common3/3.17.0-r0/git/av/src/AudioDecoder.cc
| /home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a-oe4t-linux/ignition-common3/3.17.0-r0/git/av/src/AudioDecoder.cc: In member function 'bool ignition::common::AudioDecoder::Decode(uint8_t**, unsigned int*)':
| /home/bchoi/nvidia-yocto-bsp/bchoi-build/tmp/work/armv8a-oe4t-linux/ignition-common3/3.17.0-r0/git/av/src/AudioDecoder.cc:160:33: error: 'AVCodecContext' {aka 'struct AVCodecContext'} has no member named 'ch_layout'; did you mean 'channel_layout'?
|   160 |           this->data->codecCtx->ch_layout.nb_channels;
|       |                                 ^~~~~~~~~
|       |                                 channel_layout

```

I fixed this error with my patch and hope to be checked it available.
BR,
Mark

[0002-Fix-compatibility-with-FFmpeg5.0.patch](https://github.com/gazebosim/gz-common/files/14521019/0002-Fix-compatibility-with-FFmpeg5.0.patch)

bchoineubility, 2024-03-07:
@mjcarroll I think I have no access right for this repo because I am not contributor yet. 
So, some guys can open pull reuquest with it instead of me.

And, I have an another patch for build issue, this patch can be applied to cross compile because, TRY RUN do not run correctly in the cross compile.

Can you check this patch also and include PR if no problem?

BR,
Mark

[0001-Prevent-invoking-TRY_RUN-for-cross-compiling.patch](https://github.com/gazebosim/gz-common/files/14531256/0001-Prevent-invoking-TRY_RUN-for-cross-compiling.patch)

mjcarroll, 2024-03-07:
> I think I have no access right for this repo because I am not contributor yet.

The typical github workflow is to fork this repository into your own namespace, then add the patch there as a commit and then submit a pull request back to this repository.  This does not require contributor access.

That way we can get your signoff for DCO and provide feedback.

traversaro, 2024-03-08:
> @mjcarroll I think I have no access right for this repo because I am not contributor yet. So, some guys can open pull reuquest with it instead of me.
>
> And, I have an another patch for build issue, [...]

Note that for `try_run` there is no need to apply a patch, if you are cross-compiling you can set those options via CMake options, see for example https://github.com/conda-forge/gz-common-feedstock/blob/main/recipe/build_cxx.sh#L25-L27 .

bchoineubility, 2024-03-08:
@traversaro I checked your suggestion work for me also. thanks. 

I removed my try run patch.
BR,
Mark

mjcarroll, 2024-03-11:
Closed by #581.  Thanks for reporting.