gazebosim/gz-common #614: Crash with stack trace gz::common::absPath

zhileiren, 2024-07-01:
## Environment
* OS Version: Ubuntu 22.04
* Source or binary build?
    source build
    gz-sim version: 5641ef2cc7bed13e06892d29ffd3663b2f172e76
    gz-common version: 1ffd6efa6f17d8a900bfcc93188cd3c23259101b
    built with 
    gcc version: 11.4.0 (Ubuntu 11.4.0-1ubuntu1~22.04) 
    build options: -DCMAKE_BUILD_TYPE=Coverage

## Description
* Expected behavior: Gazebo doesn't crash
* Actual behavior: Gazebo crashes
 
## Steps to reproduce:

1. run gz sim with any sdf file
2. gz service --timeout 10000 -s /gazebo/resource_paths/resolve --reptype gz.msgs.StringMsg --reqtype gz.msgs.StringMsg --req ''

## Output
[a.log](https://github.com/user-attachments/files/16047694/a.log)

zhileiren, 2024-07-02:
https://github.com/gazebosim/gz-common/blob/gz-common5/src/Filesystem.cc#L136
```c++
std::string common::absPath(const std::string &_path)
{
  return fs::absolute(_path).string();
}
```

The root cause seems to be here, no empty check for _path, or shall it be fixed in the gz-sim repository?

https://github.com/gazebosim/gz-sim/blob/gz-sim8/src/ServerPrivate.cc#L489
```c++
bool ServerPrivate::ResourcePathsResolveService(
    const msgs::StringMsg &_req,
    msgs::StringMsg &_res)
{
  // Get the request
  std::string req = _req.data();
  // return false for empty req?

  // Handle the case where the request is already a valid path
  if (common::exists(common::absPath(req)))
  {
    _res.set_data(common::absPath(req));
    return true;
  }
  ...
}
```

mjcarroll, 2024-07-17:
Thanks for bringing this up.  

It should probably be updated to use the `std::error_code` variant so that it doesn't crash but instead returns some sort of error, similar to `create_directories`

https://github.com/gazebosim/gz-common/blob/1ffd6efa6f17d8a900bfcc93188cd3c23259101b/src/Filesystem.cc#L91-L99

azeey, 2024-07-22:
@zhileiren are you interested in working on this? Can I assign it to you?

zhileiren, 2024-07-29:
Sure I'm interested. I've not contributed to the gazebo repository before, but if it's appropriate, I would like to try filing a pull request fixing this issue.

azeey, 2024-07-30:
Thanks! I've assigned the issue to you. Let me know if you need help with creating the pull request.

PR #620 (2024-08-09): Fix crash when calling absPath with empty input

# 🦟 Bug fix

Fixes #614 

## Summary

The current version of common::absPath may cause crash when calling the /gazebo/resource_paths/resolve service with empty request. By changing filesystem::absolute to the non-throwing version, and check the error_code accordingly, the crash could be avoided.

## Checklist
- [x] Signed all commits for DCO
- [x] Added tests
- [ ] Updated documentation (as needed)
- [ ] Updated migration guide (as needed)
- [ ] Consider updating Python bindings (if the library has them)
- [x] `codecheck` passed (See [contributi [... cut]
closed 2024-08-09 by iche033