gazebosim/gz-common #616: realpath issue in filesystem.cc

Kakcalu13, 2024-07-17:
Is there alternative to `realpath`? I'm using cygwin and it is progressing very good so far. It's just that I reached to the unique issue. 

See here:

```
koko@revenant ~/gz-common-ign-common3/gz-common-ign-common3/build
$ make
[  3%] Building CXX object src/CMakeFiles/ignition-common3.dir/Base64.cc.o
[  7%] Building CXX object src/CMakeFiles/ignition-common3.dir/Battery.cc.o
[ 11%] Building CXX object src/CMakeFiles/ignition-common3.dir/Console.cc.o
[ 15%] Building CXX object src/CMakeFiles/ignition-common3.dir/Filesystem.cc.o
/home/koko/gz-common-ign-common3/gz-common-ign-common3/src/Filesystem.cc: In function ‘std::string ignition::common::absPath(const string&)’:
/home/koko/gz-common-ign-common3/gz-common-ign-common3/src/Filesystem.cc:208:7: error: ‘realpath’ was not declared in this scope
  208 |   if (realpath(_path.c_str(), resolved_path) != nullptr)
      |       ^~~~~~~~
make[2]: *** [src/CMakeFiles/ignition-common3.dir/build.make:118: src/CMakeFiles/ignition-common3.dir/Filesystem.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:302: src/CMakeFiles/ignition-common3.dir/all] Error 2
make: *** [Makefile:146: all] Error 2

koko@revenant ~/gz-common-ign-common3/gz-common-ign-common3/build
```

This comes from here:
https://github.com/gazebosim/gz-common/blob/ign-common3/src/Filesystem.cc#L204

Since realpath doesn't exist in cygwin. I'm doing the most and extremely difficult challenge by install Gazebo Citadel into Windows 8.1. Installing ros2 is my next crazy challenge. 

I'm not asking for the troubleshooting or anything. I just wanted to know if there's an alternative to `realpath` since you wrote this file .

mjcarroll, 2024-07-17:
> I'm doing the most and extremely difficult challenge by install Gazebo Citadel into Windows 8.1.

Very impressive. Maybe a little insane.

Did you find this other issue that seems similar: https://github.com/avast/yaramod/issues/25#issuecomment-485226822

If you are building with cmake, you should be able to do:

```
mkdir build
cd build
cmake .. -DCMAKE_CXX_EXTENSIONS=ON
make
```

Kakcalu13, 2024-07-18:
Thanks!

It made me move forward to a new issue though! 

```
koko@revenant ~/gz-common-ign-common3/gz-common-ign-common3/build
$ make
[  1%] Building CXX object src/CMakeFiles/ignition-common3.dir/Base64.cc.o
[  2%] Building CXX object src/CMakeFiles/ignition-common3.dir/Battery.cc.o
[  3%] Building CXX object src/CMakeFiles/ignition-common3.dir/Console.cc.o
[  3%] Building CXX object src/CMakeFiles/ignition-common3.dir/Filesystem.cc.o
/home/koko/gz-common-ign-common3/gz-common-ign-common3/src/Filesystem.cc: In function ‘std::string ignition::common::absPath(const string&)’:
/home/koko/gz-common-ign-common3/gz-common-ign-common3/src/Filesystem.cc:204:25: error: ‘canonicalize_file_name’ was not declared in this scope
  204 |   char *resolved_path = canonicalize_file_name(_path.c_str());
      |                         ^~~~~~~~~~~~~~~~~~~~~~
/home/koko/gz-common-ign-common3/gz-common-ign-common3/src/Filesystem.cc:207:29: warning: declaration of ‘result’ shadows a previous local [-Wshadow]
  207 |                 std::string result(resolved_path);
      |                             ^~~~~~
/home/koko/gz-common-ign-common3/gz-common-ign-common3/src/Filesystem.cc:197:15: note: shadowed declaration is here
  197 |   std::string result;
      |               ^~~~~~
/home/koko/gz-common-ign-common3/gz-common-ign-common3/src/Filesystem.cc:213:3: error: ‘else’ without a previous ‘if’
  213 |   else if (!_path.empty())
      |   ^~~~
make[2]: *** [src/CMakeFiles/ignition-common3.dir/build.make:118: src/CMakeFiles/ignition-common3.dir/Filesystem.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1235: src/CMakeFiles/ignition-common3.dir/all] Error 2
make: *** [Makefile:146: all] Error 2

```

Seems like this is purely cygwin issue now. This is unrelated to Gazebo. I will close this as nothing we can do here. I will move this to a different repo.

Thank you so much for your time and information! Those are CERTAINLY helpful!

closed 2024-07-18 by Kakcalu13