gazebosim/gz-common #636: :farmer: `AssimpLoader.LoadBox` tests failing in gz-common-main homebrew

Crola1702, 2024-09-09:
## Environment
* OS Version: MacOS
* Gz-Ionic source build

## Description
* Expected behavior: Tests should pass
* Actual behavior: Test regressions are failing in gz-common-main-homebrew

## Output

Reference build:
* https://build.osrfoundation.org/job/gz_common-ci-main-homebrew-amd64/79/

Test regressions:
* [AssimpLoader.LoadBoxWithAnimationOutsideSkeleton](https://build.osrfoundation.org/job/gz_common-ci-main-homebrew-amd64/79/testReport/junit/(root)/AssimpLoader/LoadBoxWithAnimationOutsideSkeleton/)
* [AssimpLoader.LoadBoxInstControllerWithoutSkeleton](https://build.osrfoundation.org/job/gz_common-ci-main-homebrew-amd64/79/testReport/junit/(root)/AssimpLoader/LoadBoxInstControllerWithoutSkeleton/)
* [AssimpLoader.LoadBoxMultipleInstControllers](https://build.osrfoundation.org/job/gz_common-ci-main-homebrew-amd64/79/testReport/junit/(root)/AssimpLoader/LoadBoxMultipleInstControllers/)
* [AssimpLoader.LoadBoxNestedAnimation](https://build.osrfoundation.org/job/gz_common-ci-main-homebrew-amd64/79/testReport/junit/(root)/AssimpLoader/LoadBoxNestedAnimation/)
* [AssimpLoader.LoadBoxWithDefaultStride](https://build.osrfoundation.org/job/gz_common-ci-main-homebrew-amd64/79/testReport/junit/(root)/AssimpLoader/LoadBoxWithDefaultStride/)

All of the regressions show a problem with `mesh->TextCoordCount()` value:
```
/Users/jenkins/jenkins-agent/workspace/gz_common-ci-main-homebrew-amd64/gz-common/graphics/src/AssimpLoader_TEST.cc:401
Expected equality of these values:
  24u
    Which is: 24
  mesh->TexCoordCount()
    Which is: 0
```

There are no changes in the code, but the issue is consistent

iche033, 2024-09-09:
forward porting https://github.com/gazebosim/gz-common/pull/634 to `main` should fix homebrew CI

iche033, 2024-09-09:
did a forward port in https://github.com/gazebosim/gz-common/issues/636

Crola1702, 2024-09-16:
Closing as this is already fixed. Thanks!

closed 2024-09-16 by Crola1702